sebi2k1/node-can #6: Error while creating channel

jedcalkin, 2015-04-16:
Hi, I'm trying to get this running with:

var can = require('socketcan');
var channel = can.createRawChannel("vcan0", true);

and I get an error:

/home/jedc/node_modules/socketcan/socketcan.js:35
  return new can.RawChannel(channel, timestamps);;
         ^
Error: Error while creating channel
    at Object.exports.createRawChannel (/home/jedc/node_modules/socketcan/socketcan.js:35:10)
    at Object.<anonymous> (/home/jedc/socket_can_example.js:3:19)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:902:3

any help would be appreciated.

sebi2k1, 2015-04-16:
Please check if vcan0 is up and running (ifconfig vcan0) and make sure the can-raw kernel module is loaded as well.

jedcalkin, 2015-04-16:
Thanks I got it working by following the Virtual Interfaces section at  http://www.elinux.org/Bringing_CAN_interface_up

closed 2015-04-16 by jedcalkin