sebi2k1/node-can #7: Error during node-gyp build

drewwestrick, 2015-06-25:
I've tried executing node-gyp configure && node-gyp build on both a native raspberry pi 2 (arm v7) and via cross compilation with the same errors. I'm running the latest node & npm at versions v0.12.4 and v0.12.4 respectively built from the latest source.
```sh
pi@raspberrypi ~/node-can-master $ node-gyp build
gyp info it worked if it ends with ok
gyp info using node-gyp@2.0.1
gyp info using node@0.12.4 | linux | arm
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/pi/node-can-master/build'
  CXX(target) Release/obj.target/can/src/raw.o
In file included from ../src/raw.cc:36:0:
../src/raw.h:56:1: error: expected class-name before ‘{’ token
../src/raw.h:70:36: error: ‘Arguments’ does not name a type
../src/raw.h:70:47: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
../src/raw.h:79:44: error: ‘Arguments’ does not name a type
../src/raw.h:79:55: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
../src/raw.h:85:38: error: ‘Arguments’ does not name a type
../src/raw.h:85:49: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
../src/raw.h:91:37: error: ‘Arguments’ does not name a type
../src/raw.h:91:48: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
../src/raw.h:98:37: error: ‘Arguments’ does not name a type
../src/raw.h:98:48: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
In file included from ../src/raw.cc:36:0:
../src/raw.h:105:45: error: ‘Arguments’ does not name a type
../src/raw.h:105:56: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
../src/raw.h:112:48: error: ‘Arguments’ does not name a type
../src/raw.h:112:59: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
../src/raw.h:115:5: error: ‘uv_async_t’ has not been declared
../src/raw.h:123:5: error: ‘uv_async_t’ does not name a type
../src/raw.h: In static member function ‘static void RawChannel::async_receiver_ready_cb(int*, int)’:
../src/raw.h:115:5: error: request for member ‘data’ in ‘* handle’, which is of non-class type ‘int’
../src/raw.h:115:5: error: request for member ‘data’ in ‘* handle’, which is of non-class type ‘int’
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h: In static member function ‘static void RawChannel::Init(v8::Handle<v8::Object>)’:
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
../src/raw.cc:113:17: error: within this context
../src/raw.cc:115:58: error: no matching function for call to ‘v8::FunctionTemplate::New(v8::Handle<v8::Value> (&)(const int&))’
../src/raw.cc:115:58: note: candidate is:
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h:3455:34: note: static v8::Local<v8::FunctionTemplate> v8::FunctionTemplate::New(v8::Isolate*, v8::FunctionCallback, v8::Handle<v8::Value>, v8::Handle<v8::Signature>, int)
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h:3455:34: note:   no known conversion for argument 1 from ‘v8::Handle<v8::Value>(const int&)’ to ‘v8::Isolate*’
../src/raw.cc:117:47: error: no matching function for call to ‘v8::Persistent<v8::FunctionTemplate>::New(v8::Local<v8::FunctionTemplate>&)’
../src/raw.cc:117:47: note: candidate is:
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h:5798:4: note: static T* v8::PersistentBase<T>::New(v8::Isolate*, T*) [with T = v8::FunctionTemplate]
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h:5798:4: note:   candidate expects 2 arguments, 1 provided
../src/raw.cc:118:9: error: base operand of ‘->’ has non-pointer type ‘v8::Persistent<v8::FunctionTemplate>’
../src/raw.cc:119:9: error: base operand of ‘->’ has non-pointer type ‘v8::Persistent<v8::FunctionTemplate>’
../src/raw.cc:119:24: error: ‘NewSymbol’ is not a member of ‘v8::String’
../src/raw.cc:121:64: error: could not convert ‘RawChannel::s_ct’ from ‘v8::Persistent<v8::FunctionTemplate>’ to ‘v8::Handle<v8::FunctionTemplate>’
../src/raw.cc:122:58: error: could not convert ‘RawChannel::s_ct’ from ‘v8::Persistent<v8::FunctionTemplate>’ to ‘v8::Handle<v8::FunctionTemplate>’
../src/raw.cc:123:57: error: could not convert ‘RawChannel::s_ct’ from ‘v8::Persistent<v8::FunctionTemplate>’ to ‘v8::Handle<v8::FunctionTemplate>’
../src/raw.cc:124:57: error: could not convert ‘RawChannel::s_ct’ from ‘v8::Persistent<v8::FunctionTemplate>’ to ‘v8::Handle<v8::FunctionTemplate>’
../src/raw.cc:125:65: error: could not convert ‘RawChannel::s_ct’ from ‘v8::Persistent<v8::FunctionTemplate>’ to ‘v8::Handle<v8::FunctionTemplate>’
../src/raw.cc:126:72: error: could not convert ‘RawChannel::s_ct’ from ‘v8::Persistent<v8::FunctionTemplate>’ to ‘v8::Handle<v8::FunctionTemplate>’
../src/raw.cc:128:17: error: ‘NewSymbol’ is not a member of ‘v8::String’
../src/raw.cc:128:54: error: base operand of ‘->’ has non-pointer type ‘v8::Persistent<v8::FunctionTemplate>’
../src/raw.cc:130:42: error: ‘NODE_PSYMBOL’ was not declared in this scope
../src/raw.cc: At global scope:
../src/raw.cc:141:37: error: ‘Arguments’ does not name a type
../src/raw.cc:141:48: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h: In static member function ‘static v8::Handle<v8::Value> RawChannel::New(const int&)’:
/home/pi/.node-gyp/0.12.4/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
../src/raw.cc:143:17: error: within this context
../src/raw.cc:146:5: error: request for member ‘Length’ in ‘args’, which is of non-class type ‘const int’
../src/raw.cc:146:5: error: ‘New’ is not a member of ‘v8::String’
../src/raw.cc:146:5: error: ‘ThrowException’ was not declared in this scope
../src/raw.cc:147:5: error: invalid types ‘const int[int]’ for array subscript
../src/raw.cc:147:5: error: ‘New’ is not a member of ‘v8::String’
../src/raw.cc:147:5: error: ‘ThrowException’ was not declared in this scope
../src/raw.cc:149:5: error: ‘AsciiValue’ is not a member of ‘v8::String’
../src/raw.cc:149:24: error: expected ‘;’ before ‘ascii’
../src/raw.cc:151:14: error: request for member ‘Length’ in ‘args’, which is of non-class type ‘const int’
[... 244 lines cut ...]
../src/raw.cc: In static member function ‘static v8::Handle<v8::Value> RawChannel::Start(const int&)’:
../src/raw.cc:300:1: warning: control reaches end of non-void function [-Wreturn-type]
../src/raw.cc: In static member function ‘static v8::Handle<v8::Value> RawChannel::SetRxFilters(const int&)’:
../src/raw.cc:277:1: warning: control reaches end of non-void function [-Wreturn-type]
../src/raw.cc: In static member function ‘static v8::Handle<v8::Value> RawChannel::DisableLoopback(const int&)’:
../src/raw.cc:227:1: warning: control reaches end of non-void function [-Wreturn-type]
../src/raw.cc: In static member function ‘static v8::Handle<v8::Value> RawChannel::AddListener(const int&)’:
../src/raw.cc:194:1: warning: control reaches end of non-void function [-Wreturn-type]
../src/raw.cc: In static member function ‘static v8::Handle<v8::Value> RawChannel::New(const int&)’:
../src/raw.cc:163:1: warning: control reaches end of non-void function [-Wreturn-type]
can.target.mk:83: recipe for target 'Release/obj.target/can/src/raw.o' failed
make: *** [Release/obj.target/can/src/raw.o] Error 1
make: Leaving directory '/home/pi/node-can-master/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.18.11-v7+
gyp ERR! command "node" "/usr/local/bin/node-gyp" "build"
gyp ERR! cwd /home/pi/node-can-master
gyp ERR! node -v v0.12.4
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
```

drewwestrick, 2015-06-25:
Building with node v0.10.35 was successful. Must be some issue with a newer version of node. Might be V8 related.

sebi2k1, 2015-07-19:
Checking it with 1.12.7

sebi2k1, 2015-07-19:
Ok. In v0.12.x the C++ API changed signifanctly (see https://strongloop.com/strongblog/node-js-v0-12-c-apis-breaking/).

Will take a few days to fix it.

sebi2k1, 2015-10-03:
Hi Drew,

Daniel started porting node-can to use NAN which allows support different Node versions. I started integrated all of this in the branch belucha-merge. Feel free to give it a shot.

/Sebastian

sebi2k1, 2015-10-03:
Thanks for the work of Daniel it should now build up to NodeJS 4.1

closed 2015-10-03 by sebi2k1

drewwestrick, 2015-10-03:
Thanks for all the great work. I recently wrote some NodeJS that interfaces with socketcand. Don't know if this is a feature that could be applicable to this project. Would allow for interfacing to remote network can interfaces.

sebi2k1, 2015-10-03:
Mh, sound interesting. We could add a RawRemoteChannel or something similiar. Can you share some code?