sebi2k1/node-can #12: npm install socketcan fails

magnussp, 2015-11-18:
Hi!

While trying to install via npm install socketcan it fails for me.
I updated to latest nodejs but still didn't help.

I noticed that it during the install complained that it couldn't find the module nan. So I did a npm install nan and after that the npm install socketcan worked.

Could it be that you have forgotten to add nan in the package.json I can only see it under devDependencies and not dependencies. But I'm not to good with how these package.json files work.

sebi2k1, 2015-11-19:
Hi!

I'm going to check this on the weekend. Till now I thought in is sufficient to add nan to the devDependencies. At least this was done in the example I used.

/Sebastian

sebi2k1, 2015-11-21:
Pushed a new version which should fix the dependency issue.

closed 2015-11-21 by sebi2k1