sebi2k1/node-can #15: How to send data over CAN?

chintanp, 2016-02-04:
Can this module be used to send data over CANbus as well ? 

If yes, can you post a small example. The example in the readme isnt clear. 

Thanks

sebi2k1, 2016-02-06:
The example in README.md just logs a message received on vcan and repeats it on another vcan.

Sending is quite easy:
```javascript
var c2 = can.createRawChannel("vcan0");
	
c2.start();
	
var canmsg = { id: 10, data: new Buffer([ 0, 0xFF, 0xFE, 0xFD, 0xFC, 0xFB, 0xFA, 0xF7 ]) };
c2.send(canmsg);
```

Checkout the [documentation](https://github.com/sebi2k1/node-can/tree/master/docs) and the [tests](https://github.com/sebi2k1/node-can/tree/master/tests) for more information.

closed 2016-02-06 by sebi2k1

chintanp, 2016-02-10:
Thanks for your help. I was able to figure out how to send data. 

So I have a can message as under:
```
var canmsg = { id: 403105268, data: new Buffer([ 0, 0xFA, 0, 0x32, 0, 0, 0, 0 ]) }; 
channel.start();
setInterval(function() {
    channel.send(canmsg);
    console.log("Written to CAN device");
}, 500);```

So i get the display as "Written to CAN device". But my CAN device doesnt get actuated, so it didnt get the correct message. Upon running candump simultaneously, (which is like wireshark in this case), I see the output as under: 
```
can0  5F4   [8]  00 FA 00 32 00 00 00 00
 can0  18FF50E5   [8]  01 06 00 00 10 00 00 00
```
where the first line represents what we are sending and the next line is the CAN device's continuous status transmission. So we can see that it sends the correct data but at a **different id**. I tried sending id in canmsg in hex format as well, and I get the same results.

I tried looking inside socketcan.js but couldnt find anything that should result in such a truncation of the id. Any pointers would be appreciated. And this works from C https://gist.github.com/chintanp/2251dff8f0a0664fc02f . And complete nodejs code that doesnt yet. 
https://gist.github.com/chintanp/3c8efb788e1cb80c505d .

corn3lius, 2016-02-10:
Your thinking of the extended frame message.
https://en.wikipedia.org/wiki/CAN_bus#Data_frame

add `"ext" : true` to the canmsg

chintanp, 2016-02-10:
Yup. this works. Where was this documented ? I couldnt find it. If needed, I can write the documentation for this library, either as readme or github-pages. 

Thanks again. :+1:

sebi2k1, 2016-02-14:
@chintanp Please feel free to enhance README.md.