sebi2k1/node-can #21: How to decode all signals in a message at once?

daixtrose, 2016-08-03:
I want to decode *all* signals of a message at once based on the kcd information, something along these lines:
```javascript
"use strict";

var can = require('socketcan');
var fs = require('fs');

// Parse database
var network = can.parseNetworkDescription("GWI_TT.kcd");
var channel = can.createRawChannel("can0");

var db      = new can.DatabaseService(channel, network.buses["Private"]);

channel.addListener("onMessage", function(msg) {
    console.log(msg);

    let message_info = db.messages[msg.id];
    
    if (message_info.id == '3') 
    {
	for (let signal_name in message_info.signals)
	{
	    let signal_info = message_info.signals[signal_name];

	    // decode that value based on information in signal_info
	    let value = ???
	}
    }
} );

channel.start();
``` 

Any hints appreciated.

daixtrose, 2016-08-03:
I refine my question: Is this code here guaranteed to print the actual values?

```javascript
"use strict";

var can = require('socketcan');
var fs = require('fs');

// Parse database
var network = can.parseNetworkDescription("GWI_TT.kcd");
var channel = can.createRawChannel("can0");

var db      = new can.DatabaseService(channel, network.buses["Private"]);

channel.addListener("onMessage", function(msg) {
    console.log(msg);

    let message_info = db.messages[msg.id];
    
    if (message_info.id == '3') 
    {
	// console.log(db.messages[message_info.name]);

	let db_messages = db.messages[message_info.name];
	
	for (let signal_name in message_info.signals)
	{
	    let db_signal = db_messages.signals[signal_name];
	    
	    console.log(db_signal.name + ": " + db_signal.value);
	}
    }
} );
```

sebi2k1, 2016-08-03:
On new message the database service is called first (database registers its own listener first), which will decode all the signals and stores the latest value in the database. After that the listener onMessage is called which prints the latest stored signal values.

So yes, the provided code will print the actual values.

corn3lius, 2016-08-03:
yes the translated (slope/offset) values. 

But the values could lag the message reception. since you are duplicating this method 
https://github.com/sebi2k1/node-can/blob/master/socketcan.js#L236
and the signal decoding happens at the end of that function.

What about adding an event for message reception at the end of that function ?

```

util.inherits(DatabaseService, events.EventEmitter);
DatabaseService.prototype.onMessage = function (msg) {
	id = msg.id | (msg.ext ? 1 : 0) << 31;

	var m = this.messages[id];
        ... 
	for (i in m.signals) {
	       ... 
		s.update(val);
	}
        this.emit(id, m);
}

```
 and in your code 

```
db.on([msg_id], function(message){ 
   // handle message }
) 

```

sebi2k1, 2016-08-03:
@corn3lius I like the extensions in terms of making it easy for the user to get an notification when a message was received and processed by db service.

One thing I dont understand is your statement:
> But the values could lag the message reception.
If db service process events first and afterwards the message handlers are called, why can values lag the message reception?

@daixtrose I can't test the fix @corn3lius propose right now, if you like the idea please try it out and provide feedback.

daixtrose, 2016-08-04:
Please help me. i do not see why @corn3lius solution is an advantage. AFAICS it only adds the message id  to the event. I would rather prefer to extend the library such that it provides an RXJS compatible observable. Then I can apply a `Where` filter based on message id.

corn3lius, 2016-08-04:
Ran some tests: as long as the onMessage listeners is synchronous where `db.onMessage`  happens before we define our own  `channel.onMessage` we will be ok. 

ie: 
```
     var channel = can.createRawChannel("vcan0");

     channel.addListener("onMessage", function(msg){ 
        // signals would not have been processed yet. (previous values and not msg values ) 
        var sigs = db.messages["TestMessage"].signals;  
     });
     var db      = new can.DatabaseService(channel, network.buses["Motor"]);
    
     channel.addListener("onMessage", function(msg){ 
        // signals would have been processed in db
        var sigs = db.messages["TestMessage"].signals; 
     });
```

sebi2k1, 2016-08-04:
@corn3lius confirmed requested behavior.

closed 2016-08-04 by sebi2k1