sebi2k1/node-can #22: Signal units not correctly parsed?

daixtrose, 2016-08-03:
```javascript
var can = require('socketcan');
var network = can.parseNetworkDescription("node_modules/socketcan/samples/can_definition_sample.kcd");
console.log(JSON.stringify(network, null, '  '));
```

yields the following output
```
                ....
                {
                  "name": "SpeedKm",
                  "bitLength": 24,
                  "endianess": "little",
                  "labels": {},
                  "slope": 0.2,
                  "intercept": 0,
                  "units": "",
                  "type": "unsigned",
                  "defaultValue": 0,
                  "bitOffset": 30
                },
               ...
```

but the kcd file contains a unit 
```xml
	       <Signal name="SpeedKm" offset="30" length="24" endianess="little">
	       	<Notes>Middle speed of front wheels in kilometers per hour.</Notes>
	       	<Consumer>
	       		<NodeRef id="12"/>
	       		<NodeRef id="16"/>
	       	</Consumer>	       	
	       	<Value slope="0.2" intercept="0" unit="km/h"/>
	       	<LabelSet>
	       		<Label type="error" name="invalid" value="16777215"/>
	       	</LabelSet>
	       </Signal>
```

daixtrose, 2016-08-03:
Maybe parse_kcd.js should read
```javascript
								_s.units = value[0]['$'].unit ? value[0]['$'].unit : "";
```
instead of 
```javascript
								_s.units = value[0]['$'].units ? value[0]['$'].units : "";
```
?

daixtrose, 2016-08-03:
These changes in `parse_kcd.js` solve it for me, please confirm:
```
156c156
< 								_s.units = value[0]['$'].units ? value[0]['$'].units : "";
---
> 								_s.units = value[0]['$'].unit ? value[0]['$'].unit : "";
200c200
< 						_s.units = value[0]['$'].units ? value[0]['$'].units : "";
---
> 						_s.units = value[0]['$'].unit ? value[0]['$'].unit : "";
```

sebi2k1, 2016-08-03:
Looks good! Just cross checked with [KCD definitions](https://github.com/dschanoeh/Kayak/blob/master/Kayak-kcd/src/main/resources/com/github/kayak/canio/kcd/loader/Definition.xsd)

sebi2k1, 2016-08-03:
Fixed in f36419c358fccffb7e9615c76891170ba41a3e77.

closed 2016-08-03 by sebi2k1