sebi2k1/node-can #23: TypeError: Cannot read property 'NetworkDefinition' of undefined

zdog1230, 2016-08-03:
I've been getting this error and I'm stuck to see why. I've attached my definitions file below too so you can see the formatting. Thanks

events.js:160
      throw er; // Unhandled 'error' event
      ^

TypeError: Cannot read property 'NetworkDefinition' of undefined
`<NetworkDefinition>
	<Node id="4" name="Main Encoder"/>
    <Node id="6" name="Wing Instrumentation"/>
    <Node id="7" name="Tower Top"/>
	<Node id="8" name="Wind Sensor"/>
	<Node id="10" name="Safety Brake"/>
	<Node id="14" name="UPS Control"/>
	<Node id="11" name="Servo Drive 1"/>
	<Node id="12" name="Servo Drive 2"/>
	<Node id="13" name="Servo Drive 3"/>
	<Node id="9" name="Inverter Control"/>
	<Node id="1" name="Master 1"/>
	
	<Bus name="Turbine">
		<Message id="0x6050h" name="Wind">
			<Consumer>
				<NodeRef id="8"/>
			</Consumer>
			<Signal name="SpeedMPH" offset="0"/></Signal>
	</Message>	
	</Bus>	
	
</NetworkDefinition>

`

sebi2k1, 2016-08-04:
Don't you have a longer backtrack?

Please note, message identifiers must be written in plain hex format, trailing 'h' is not allowed.

corn3lius, 2016-08-04:
```
            <Signal name="SpeedMPH" offset="0"/></Signal>
```

is not valid

closed 2017-02-25 by sebi2k1