sebi2k1/node-can #24: How to create obervables using channel.addListener?

daixtrose, 2016-08-04:
Hi!

I would like to create observables for `channel.addListener('onMessage', cb)`. Do you have some example code? Mine failed silently

```javascript
var observableMessages = Rx.Observable.fromEvent(channel.onMessage);

var subscription = observableMessages.subscribe((msg) => {
    let valuesTimeSlice = createValuesTimeSlice(msg);

    console.log(JSON.stringify(valuesTimeSlice, null, '  '));
    io.emit('valuesTimeSlice', valuesTimeSlice);
});
````

sebi2k1, 2016-08-04:
Sorry, don't have a clue. I leave it open, maybe somebody else can help.

daixtrose, 2016-08-05:
My current solution is (using latest RxJS version):

```javascript
var Rx = require('rxjs/Rx');
...
var observableMessages = Rx.Observable.create(obs => {
    let cb = function(msg) { obs.next(msg); };
    channel.addListener("onMessage", cb);
    return function() {
	// See #25
	// channel.removeListener(cb);
    }
});

var subscription = observableMessages.subscribe((msg) => {
    console.log(msg);
    ...
});

```

daixtrose, 2016-08-05:
Note that this solution is not compatible with Rx rules as long as `removeListener` is unavailable. If someone calls `subscription.dispose()` the events will still get through, which is disallowed behaviour.

There are 2 possibilities to cure this:
- refine the above solution to maintain state and disable `obs.netx(msg)` in case of an unsubscription
- add `removeListener` as method to `RawChannel`

daixtrose, 2016-08-05:
Current conforming solution:

```javascript
var observableMessages = Rx.Observable.create(obs => {
    let active = true;
    let cb = function(msg) {
	if (active) {
	    obs.next(msg);
	}
    };
    
    channel.addListener("onMessage", cb);

    return function() {
	active = false;
    }
});
```

sebi2k1, 2016-08-05:
Ok, we will add removeListener.

closed 2016-08-05 by sebi2k1

daixtrose, 2016-09-29:
I'd prefer to keep the issue open until the method is added.