sebi2k1/node-can #26: Network goes down?

nivoset, 2016-11-18:
I've been using this library successfully for a while, but recently ive had some odd occurances, if i dont run the node app, the can network is fine, however when running the app, i will suddenly 'kill' my network off, and to get it up i need to do a ifconfig can0 down, can0 up.

When this happens, my software does record can messages that shouldn't exsist on the network, (id 4 and 40 normally) these are not used.

Has anyone else had this problem?`channel.addListener("onMessage", function (msg) {
	var id = msg.id;
	var message = msg.data.toString('hex');
	var datagram = 0;
	var source = 0;
	var destination = 0;
	switch (id) {
	case 1175421952: /* code removed */
		break;
	case 1175420928: /* code removed */
		break;
	case 1175422976: /* code removed */
		break;
	default:
		console.log("unknown id: " + id + "\t hex: " + id.toString(16));
                console.log(msg);
		break;
	}
});
channel.start();`

I'm not sure why this is going on, and it worked until today fine (no non expected messages or anything)

sebi2k1, 2016-11-19:
node-can doesn't generate can messages on its own (we are also not using bcm module), so I expect this is somehow generate by another CAN node.

Can you confirm the issue is still existent if you end you application? You may have to use a separate application just for monitoring (e.g. candump).

Which hardware are you using?

nivoset, 2016-11-19:
I am using this cape:
http://www.ebay.com/itm/like/261692397694?lpid=82&chn=ps&ul_noapp=true
Can Cape from waveshare

And no, if the app isn't running, it doesn't go down and i do not see the error frames. (using candump)

I don't see it at home when i've been using this, so looking into noise where i have it wired up, but it was working fine until yesterday. 

I'm also going to install/compile the drivers. maybe something happened on this unit when i installed them.

sebi2k1, 2016-11-19:
Sorry, you are seeing error frames with candump? Error frames are generated by the CAN controllers on the network on bitrate issues, id collision or other communication issues. Did you check if the bus is properly terminated (120ohm each on end and beginning) and all nodes running the exact same bit rate?

nivoset, 2016-11-19:
no, there are no error frames in candump. and i see can messages (in socketcan) that do not appear in candump as well (id's that are not on the network)

nivoset, 2017-06-20:
Ive had this problem before, and fixing terminations or me helped out. 
otherwise i added some script to look for communication in x timeframe and if not then i perform a reset of the can network. (this was to add in case it is ever needed, so far it hasn't been called since i fixed termination)

nivoset, 2017-06-20:
so it was a termination thing, just didn't happen often enough to look like it.. I also added some code to look for the can to die out (error) or no communication in 20 seconds and have it reset it. but neither of those things has happened since i changed the terminations (though the new setup isn't at either side, that is how it was... still trying to figure it out)

closed 2017-06-20 by nivoset