sebi2k1/node-can #28: Message update on no value signals

Flycode-1, 2017-02-07:
Hi !

**Following the example to update and send a message :** 

```
// Update tank temperature
db.messages["TankController"].signals["TankTemperature"].update(80);

// Trigger sending this message
db.send("TankController");
```

**How could I send a message that as no value that would be defined such as the following in a kcd file :** 

```
<Message id="0x00A" name="Airbag">
		 	<Producer>
		 		<NodeRef id="2"/>
		 	</Producer>
		 	<Signal name="DriverAirbagFired" offset="0"/>
		 	<Signal name="CodriverAirbagFired" offset="1"/>
		 	<Signal name="DriverSeatOccupied" offset="2"/>
		 	<Signal name="CodriverSeatOccupied" offset="3"/>
		 	<Signal name="DriverSeatbeltLocked" offset="4"/>
		 	<Signal name="CodriverSeatbeltLocked" offset="5"/>
		 	<Signal name="AirbagConfiguration" offset="8" length="8"/>
		 	<Signal name="SeatConfiguration" offset="16" length="8"/>
		 </Message>
```

How to update and send such message if i want to send for example the signal "DriverSeatOccupied" ?

Many thanks for your help!

sebi2k1, 2017-02-08:
I don't get the question, you can send a message defined in the database via db.send, in your case: 
db.messages["Airbag"].signals["DriverSeatOccupied"].update(xx);
db.send("Airbag")

Flycode-1, 2017-02-08:
Hi Sebastian thank you for you quick response.

I am glad to know this is something possible.What is the xx parameter in
send in this case ?

db.messages["Airbag"].signals["DriverSeatOccupied"].update(*xx*);
db.send("Airbag")

kind regards,

sebi2k1, 2017-02-09:
XX is the value of the signal in the message.

corn3lius, 2017-02-09:
FYI : 

https://github.com/sebi2k1/node-can/blob/master/parse_kcd.js#L178

the length on your signal is defaulted to 1 bit so the only values you can code are ( 0|1)

Flycode-1, 2017-02-09:
Thanks for your help , this works perfectly. 
Again congratulation for this NodeJS SocketCAN extension, it is a very handy extension !

closed 2017-02-25 by sebi2k1