sebi2k1/node-can #30: Request: onClose/onError handlers

fabdrol, 2017-03-09:
`node-can` doesn't seem to have error and close handlers (unless I'm missing something). It'd be nice to provide this in the module, allowing us to do something if the connection to the bus fails or closes. 

When creating the channel, this can be caught using a `try..catch`, but how would I go about doing after we have a working channel?

sebi2k1, 2017-03-09:
In CAN the connection to the bus doesn't really fail, the CAN controller will encounter error frames until it goes bus off. But this is already possible today by tracking the the error frames and error state.

The only use case I can think of might be the case when the network devices get suddenly down or is being removed, but this seems like a rare situation.

Which situation are you trying to solve exactly?

fabdrol, 2017-03-09:
Exactly that kind of situation, let's say the USB can device is removed - in that case we'd like to be able to tell the user what's happening.

gautaz, 2017-03-31:
Same situation here, same needs.

It would also be useful to handle some sort of reconnection algorithm but that could be considered as a second step on the way to handle this type of issues.

sebi2k1, 2017-04-05:
Working on it.

sebi2k1, 2017-04-14:
Fixed with commit f893b8286194a3db440b62c3a2b769d2c1be1f30.

closed 2017-04-14 by sebi2k1

gautaz, 2017-04-15:
Great !
I will try this as soon as I get back to work.

So we have to listen for the `onStopped` event. What is the best way to reinitiate a connection to the can bus? Is creating a new raw channel the best option or can we reuse the old one?

sebi2k1, 2017-04-15:
Correct, listen for onStopped. The channel is properly brought down, so you can just start it again.

gautaz, 2017-04-20:
Still working on it, I'm using a [CANUSB](http://www.can232.com/?page_id=16) module and I have currently the following behaviors:

- if I am disconnecting the CAN serial line from the module, no `onStopped` event is send but I still need to investigate if anything is reported by this module on the USB side in this case.
- if I am disconnecting the USB interface, an `onStopped` event is generated but again I have another issue with the device USB reconnection (not related to `node-can` but perhaps to [slcand](https://github.com/linux-can/can-utils) which seems unable to bring back the connection with the device).

What happens to `node-can` if the `slcan0` network interface is temporarily not present?
The question arises because this interface might temporarily disappear during USB disconnection/reconnection (udev will [remove the interface and bring it back again](http://pascal-walter.blogspot.fr/2015/08/installing-lawicel-canusb-on-linux.html)).

sebi2k1, 2017-04-20:
The onStopped event is generated whenever the CAN network device goes down or is destroyed. So to answer your question, if the network device is temporarily not present (=network interface is down or doesn't exit anymore) the onStopped event should be generated and the channel is stopped.

fabdrol, 2018-05-16:
@sebi2k1 I missed this conversation, great work, thanks! One more (related) question: with candump we can catch error frames as if they are normal messages - and can we act on them?