sebi2k1/node-can #38: NODE_MODULE_VERSION mismatch

lostdesign, 2017-08-23:
I tried using electron and socketcan/node-can. When ever i try to run the app, that following error message appears:

```
A JavaScript error occurred in the main process
Uncaught Exception:
Error: The module '/home/weller/Documents/01_CODING/myProject/node_modules/rawcan/build/Release/can_wrap.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (/home/user/Documents/01_CODING/myProject/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/user/Documents/01_CODING/myProject/node_modules/rawcan/dist/can_wrap.js:2:18)
```

No matter what node version i am using - there will always be a mismatch and i have no clue how to fix it. What i have tried so far:

- removing all `node_modules` and reinstalling
- rebuilding `npm rebuild`
- swapping node versions
- reinstalling node completely

sebi2k1, 2017-10-07:
You are not using node-can but rawcan (different implementation).

lostdesign, 2017-10-10:
How come? I used  `npm install socketcan` to install the package?

sebi2k1, 2017-10-10:
Cause the build error doesn't reference socketcan but a module called rawcan:
    at Object.<anonymous> (/home/user/Documents/01_CODING/myProject/node_modules/**rawcan**/dist/can_wrap.js:2:18)

closed 2017-10-10 by sebi2k1

lostdesign, 2017-10-11:
I will try to install it again and check what happens then. Thanks