sebi2k1/node-can #42: 'onStopped' is not invoked

curisiro, 2018-03-28:
Hi, the example code as below:
After 1 second timeout, I can see 'timeout' log in the console. But I cannot see 'stop' log in the console, it means 'onStopped' isn't invoked. 

Can anyone help me to check and fix this?

var can = require('socketcan');
var channel= can.createRawChannel("can1");

channel.addListener("onStopped", function() {
	console.log('stop');
});

channel.start();

setTimeout(function() {
	console.log('timeout');
		channel.stop();
}, 1000);

swapsCAPS, 2019-02-09:
It is only invoked on `iface` `down`

juleq, 2019-02-09:
onIfaceDown might have been the original purpose and onStopped is just a misleading name? It would make more sense to me since I do not need an event to tell me that I have just called the stop method.

What exactly is you use case for the onStopped listener? 

I think you should probably consider wrapping the stop method, inherit from event emitter and emit a custom stop event if you want to give other parts of your application a notification.

Further, if you are interested in the bus error states: If a received message has an err property, it contains error frame information that you can use to determine the state of the bus.

sebi2k1, 2020-01-08:
Fixed with #78

PR #78 (2020-01-08): Call onStopped on Stop() too
Fixes #42
closed 2020-01-08 by sebi2k1