sebi2k1/node-can #43: node-can fails to build on node v10

fabdrol, 2018-06-15:
I recently upgraded to node v10. When building the module, a build error occurs that seems to be related to changes in v8. 

```
> socketcan@2.2.2 install /home/decipher/Apps/essense-control-api/node_modules/socketcan
> node-gyp rebuild

make: Entering directory '/home/decipher/Apps/essense-control-api/node_modules/socketcan/build'
  CXX(target) Release/obj.target/can/src/rawchannel.o
../src/rawchannel.cc: In member function 'void RawChannel::async_channel_stopped(int)':
../src/rawchannel.cc:485:32: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
           callback.Call(0, NULL);
                                ^
In file included from ../src/rawchannel.cc:23:0:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^
../src/rawchannel.cc:487:60: warning: 'v8::Local<v8::Value> Nan::Callback::Call(v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
           callback.Call(Nan::New(listener->handle), 0, NULL);
                                                            ^
In file included from ../src/rawchannel.cc:23:0:
../../nan/nan.h:1596:3: note: declared here
   Call(v8::Local<v8::Object> target
   ^
../src/rawchannel.cc: In member function 'void RawChannel::async_receiver_ready(int)':
../src/rawchannel.cc:561:32: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
           callback.Call(1, argv);
                                ^
In file included from ../src/rawchannel.cc:23:0:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^
../src/rawchannel.cc:563:60: warning: 'v8::Local<v8::Value> Nan::Callback::Call(v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
           callback.Call(Nan::New(listener->handle), 1, argv);
                                                            ^
In file included from ../src/rawchannel.cc:23:0:
../../nan/nan.h:1596:3: note: declared here
   Call(v8::Local<v8::Object> target
   ^
  SOLINK_MODULE(target) Release/obj.target/can.node
  COPY Release/can.node
  CXX(target) Release/obj.target/can_signals/src/signals.o
../src/signals.cc: In function 'Nan::NAN_METHOD_RETURN_TYPE DecodeSignal(Nan::NAN_METHOD_ARGS_TYPE)':
../src/signals.cc:111:35: error: no matching function for call to 'v8::Value::ToUint32()'
     offset    = info[1]->ToUint32()->Uint32Value();
                                   ^
In file included from /home/decipher/.node-gyp/10.4.1/include/node/node.h:63:0,
                 from ../../nan/nan.h:51,
                 from ../src/signals.cc:18:
/home/decipher/.node-gyp/10.4.1/include/node/v8.h:2400:44: note: candidate: v8::MaybeLocal<v8::Uint32> v8::Value::ToUint32(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Uint32> ToUint32(
                                            ^
/home/decipher/.node-gyp/10.4.1/include/node/v8.h:2400:44: note:   candidate expects 1 argument, 0 provided
../src/signals.cc:112:35: error: no matching function for call to 'v8::Value::ToUint32()'
     bitLength = info[2]->ToUint32()->Uint32Value();
                                   ^
In file included from /home/decipher/.node-gyp/10.4.1/include/node/node.h:63:0,
                 from ../../nan/nan.h:51,
                 from ../src/signals.cc:18:
/home/decipher/.node-gyp/10.4.1/include/node/v8.h:2400:44: note: candidate: v8::MaybeLocal<v8::Uint32> v8::Value::ToUint32(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Uint32> ToUint32(
                                            ^
/home/decipher/.node-gyp/10.4.1/include/node/v8.h:2400:44: note:   candidate expects 1 argument, 0 provided
../src/signals.cc: In function 'Nan::NAN_METHOD_RETURN_TYPE EncodeSignal(Nan::NAN_METHOD_ARGS_TYPE)':
../src/signals.cc:212:32: error: no matching function for call to 'v8::Value::ToUint32()'
     offset = info[1]->ToUint32()->Uint32Value();
                                ^
In file included from /home/decipher/.node-gyp/10.4.1/include/node/node.h:63:0,
                 from ../../nan/nan.h:51,
                 from ../src/signals.cc:18:
/home/decipher/.node-gyp/10.4.1/include/node/v8.h:2400:44: note: candidate: v8::MaybeLocal<v8::Uint32> v8::Value::ToUint32(v8::Local<v8::Context>) const
[... 64 lines cut ...]
                                                ^
can_signals.target.mk:95: recipe for target 'Release/obj.target/can_signals/src/signals.o' failed
make: *** [Release/obj.target/can_signals/src/signals.o] Error 1
make: Leaving directory '/home/decipher/Apps/essense-control-api/node_modules/socketcan/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.13.0-45-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/decipher/Apps/essense-control-api/node_modules/socketcan
gyp ERR! node -v v10.4.1
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN ajv-keywords@3.2.0 requires a peer of ajv@^6.0.0 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! socketcan@2.2.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the socketcan@2.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

selimdantesinferno, 2018-08-28:
HI parkertomatoes i am dealing with this issue on socketcan i don't know how to fix it

Error: Cannot open /my-project/node_modules/socketcan/build/Release/can.node: Error: The module '/my-project/node_modules/socketcan/build/Release/can.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
Thanks in advance for your response.

parkertomatoes, 2018-09-05:
Hi, @selimdantesinferno - did you get that from trying to build with my pull request or just in general?

Sometimes I get that in general if I change node versions using nvm or something, and some binary dependency was built using the other node install. I'm not really a node expert, so this is probably the wrong way, but my inelegant workaround is usually to delete the node_modules directory and re-run npm install.

selimdantesinferno, 2018-09-06:
Hi it worked finally it was a problem with the version of electron 1.6

sebi2k1, 2018-09-09:
Deprecated API remove by @parkertomatoes and publish in 2.3.0. Please use that version.

closed 2018-09-09 by sebi2k1