sebi2k1/node-can #47: Unable to install on Windows (target system is Linux)

ralmo, 2018-10-16:
Hey guys,

I'd really appreciate some help in the following regard.

We're developing on Windows machines but our target will be a small embedded Linux. Communication should use CAN and that's how I stumpled upon this very promising package.

I intended to add a conditional switch in my application: use some sort of CAN-dummy-mock during development on Windows then build it via Webpack and deploy it to Linux where `socketcan` can be used.

However, when I try to install it using `yarn add socketcan` I receive the following error:
```
info fsevents@1.2.4: The platform "win32" is incompatible with this module.
info "fsevents@1.2.4" is an optional dependency and failed compatibility check. Excluding it from installation.
error socketcan@2.3.0: The platform "win32" is incompatible with this module.
error Found incompatible module
```

I understand that this is due to fsevents needing Linux/OSx to work but is there a way to get it to install anyway and only make use of it on the target system?

Best regards,
Rolf

parkertomatoes, 2018-10-18:
I get around this by adding linux-only dependencies like socketcan to "optionalDependencies" in package.json instead of "dependencies"

ralmo, 2018-10-22:
@parkertomatoes : that would allow me to at least link it in my `package.json`, thanks.

But if I don't download it locally (because "optional") how would I later deploy it on my target? 

Apparently, my target environment (ARM Linux, 32 bit) has some limitations and I cannot make use of `npm install` to install it there. Something about a missing compiler toolchain.

Normally I would bundle everything with Webpack and deploy it then (if necessary by simply copying some files that cannot be parsed/bundled).

parkertomatoes, 2018-10-24:
The tricky part about that is getting the cross-compile environment for your native binaries. 

You could try using Docker or WSL to give yourself a Linux-like environment. Assuming your architecture is armhf (Raspberry Pi's), install the cross-compile gcc tools (e.g. "sudo apt-get install gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf" ).

After that, "env CC=arm-linux-gnueabihf-gcc CXX=arm-linux-gnueabihf-g++ npm install --build-from-source --arch=armhf"

If that doesn't work, all I can really suggest is running "npm install" on the target as part of your deployment process.

ralmo, 2018-10-25:
Thanks for your advice.

If I understand you correctly, the following steps are necessary on a Linux pc:
1. ) Install the fitting cross-compiler
2.) Temporarily rewire environment variable CC to the proper cross compiler
3.) Execute `npm install` with the `--arch` flag

Apparently that doesn't require me to do changes to a socketcan fork meaning it is very maintainable. 
Nice :)

Until now I managed to manually cross-compile the cc-files for the target plattform by experimenting with node-pre-gyp (which also needs a cross-compiler and changes my local socketcan-repository).

closed 2018-10-31 by ralmo