sebi2k1/node-can #48: Excessive cpu usage?

sliem, 2018-12-05:
First of all, thank you for writing this library!

Processing 10k CAN messages per second takes about 40% of the CPU on my fairly beefy laptop ( i7-4710MQ CPU @ 2.50GHz), while the equivalent golang and rust programs take about ~4.5% cpu.
These are numbers reported by htop.

I do expect rust and golang to be faster, but as socketcan is written in C++ so not that much faster. Profiling the code revealed that most of the time (95%) it is in epoll_pwait waiting for work. This might be a lie by the node profiler however, as presumably the process is simply waiting for work.

My question is: Is this the expected overhead from libuv? Or something tunable? Like the poll timeout (currently 100 ms)?

Below are the results of my testing, I've tried to elide uninteresting details, if I elided something useful I'm happy to provide them.

### Test scenario (node v10.15.1, ubuntu 16.04, socketcan 2.3.0)

Writing about 10k can msgs per second to the `vcan0` using `cangen`. The test program `canjs.js` simply just reads those can messages with a trivial handler. 

```javascript
$ cat canjs.js
var socketcan = require("socketcan");

cs = socketcan.createRawChannel("vcan0", true);

console.log("canjs connected");

cs.addListener("onMessage", function(msg) {
  // console.log(msg.id);
  var a = msg.id;
});

cs.start();
```

Profiling `canjs.js` reveals that major of the ticks are spent in `epoll_pwait`

```bash
$ node --prof canjs.js 
canjs connected
[wait few seconds]
^C 
$ node --prof-process isolate-0x*.log
[...]
 [Summary]:
   ticks  total  nonlib   name
     21    0.1%    0.1%  JavaScript
  16430   99.4%   99.8%  C++
     15    0.1%    0.1%  GC
     63    0.4%          Shared libraries
     11    0.1%          Unaccounted
[...]
 [C++]:
   ticks  total  nonlib   name
  15682   94.9%   95.3%  epoll_pwait
    140    0.8%    0.9%  __libc_recv
     60    0.4%    0.4%  ioctl
[...]
```

Using `strace` I found that `epoll_pwait` is called everytime `async_receiver_ready` finish exhausting the available messages on the socket. Presumably by libuv, as the `uv_loop` is epoll-based on linux. 

```
$ strace -rqp [pid]   # showing the single thread as the multi-thread is just harder to follow.
[...]
     0.000029 recvfrom(20, "\202\6\0\0\1\0\0\0`\0\0\0\0\0\0\0", 16, MSG_DONTWAIT, NULL, NULL) = 16
     0.000027 ioctl(20, SIOCGSTAMP, 0x7ffd3630af00) = 0
     0.000030 recvfrom(20, "\322\2\0\0\6\0\0\0U3v3z\216\0\0", 16, MSG_DONTWAIT, NULL, NULL) = 16
     0.000028 ioctl(20, SIOCGSTAMP, 0x7ffd3630af00) = 0
     0.000027 recvfrom(20, 0x7ffd3630af80, 16, MSG_DONTWAIT, NULL, NULL) = -1 EAGAIN (Resource temporarily unavailable)
     0.000022 epoll_pwait(3, [{EPOLLIN, {u32=8, u64=8}}], 1024, -1, NULL, 8) = 1
[...]
```

If we summarize the syscalls made `epoll_pwait` falls in the ranking. The futex'es are seemingly only during the start-up.

```
$ strace -cfq node canjs.js
canjs connected
[wait few seconds]
^C 
% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 80.01   19.304016       53922       358        17 futex
 10.63    2.565721          12    221348           poll
  2.96    0.714898           5    141162     33658 recvfrom
  2.76    0.665723           6    107536        10 ioctl
  1.77    0.426898          13     33659           epoll_pwait
  1.08    0.260565           8     33663           write
  0.76    0.183782           5     33702           read
  0.01    0.001323          12       114           brk
  0.00    0.000992           1       842           mprotect
  0.00    0.000526           8        67           mmap
  0.00    0.000242           3        72        20 open
  0.00    0.000202           6        33           munmap
  0.00    0.000183           5        35           rt_sigaction
  0.00    0.000155           1       140        65 stat
  0.00    0.000107          10        11        11 access
[...]
```

sliem, 2018-12-06:
I was able to reduce the CPU usage by reducing the number of `poll` calls with a `usleep`. As far as I understand the polling loop keeps running, as fast as possible, even when the callback is already handling data. Sleeping after each `poll` calms the loop down at the cost of some awkward tuning.

```C
  void ThreadEntry()
  {
    struct pollfd pfd;

    pfd.fd = m_SocketFd;
    pfd.events = POLLIN|POLLHUP|POLLERR;

    while (!m_ThreadStopRequested)
    {
      pfd.revents = 0;

      if (likely(poll(&pfd, 1, 100) >= 0))
      {
        if (likely(pfd.revents & POLLIN))
          uv_async_send(&m_AsyncReceiverReady);

        if (pfd.revents & (POLLHUP|POLLERR))
        {
          uv_async_send(&m_AsyncChannelStopped);
          break;
        }
        usleep(10000);  // NEW sleep for 10 ms to reduce polling.
      }
      else
      {
        break;
      }
    }
  }
```

sliem, 2018-12-06:
I wrote a rough patch which adds an option to createRawChannel for sleeping in the poll loop. See #https://github.com/sliem/node-can/commit/7e8e166b0ee4e08e584e7bfac4656f35a3ec0394. I can create an oull request, if this is an acceptable approach?

juleq, 2018-12-06:
@sliem: Nice observations. I have just finished an application that streams 200 decoded signals (total) from two CAN busses to file. Frame rate is about 400fps per bus. Processor is a Cortex A9@1GHz. CPU usage is about 27% (top).

Have you seen the rawcan npm package? I do not know if that is under active development (repo on Github seems to be gone, forks are available). Here is what the author said contrasting his approach with the one of @sebi2k1 :

>  The main difference is that node-can is not evented- sends are synchronous and receives are done on a separate thread. By contrast, this package ties into the libuv based node event system, which uses epoll under the hood.

I like the approach of @sebi2k1 to include encoding/decoding, so I use his. But I also wonder what performance gains one could expect from transitioning to the libuv approach under the hood.

I think your proposal is a valid quick fix if you do not care about latency. Have you tried what you will gain when you lower the time to 1000us or even below? Some numbers would be interesting.

It would also be nice to know @sebi2k1 take on the libuv approach.

@sebi2k1 Thank you very much, sir. This is a good one.

sliem, 2018-12-07:
@sebi2k1 that does makes sense. It was that hypothesis that made me try the sleep. My sleepy hack is of course just a hack, but I couldn't think of way to introduce (non-blocking) synchronization between the poll thread and uv-loop without another poll which would defeat the purpose in the first place. I'm not very versed in threads however.

@juleq looked at rawcan briefly, but it's state of almost non-existence made me skip it.
It's true that my particular application isn't very latency sensitive, but note that any latency introduced isn't very linear as the poll triggers a callback `async_receiver_ready` that reads the available (up to `MAX_FRAMES_PER_ASYNC_EVENT = 100`) messages as fast as it cans.

I've tested some different sleep durations and their impact on `CPU%`. To (very) roughly estimate the impact on latency I calculate read frames per second by counting the number of frames read for ten seconds. These specific number are all likely very machine dependent, but the behavior should be the same.

| sleep time (µs) | CPU% (htop) | frames per second |
| - | - | - | 
| 0 | ~44% | 5733 |
| 100 | ~22% | 5732 |
| 1000 | ~19% | 5739 |
| 10000 | ~7.5% | 5737 |
| 100000 | ~2% | 1001 |

In the last scenario the code polls every 0.1 second, triggering the callback which read 100 frames before terminating. Which limiting the throughput to ~1000 fps.

sebi2k1, 2018-12-08:
You may give commit 618e819f3a59295dc0834f1d937b3852eb78be52 a try, it introduces a mutex/cond to block the poll thread in a condition until the even was handled.

juleq, 2018-12-10:
@sebi2k1: Thanks, looks promising. I have installed it below my aforementioned application. CPU usage is down from 27-30% to 14-18%.

Background: I use the OnChange listener in my application and I use a Vector config to generate a sine for every signal on two CAN busses (each 20% load @250k). I record 100 decoded signals/bus.

I have not done a thorough evaluation (e.g. checking for dropped frames, etc.) but it has been running for a few hours now.

@sliem: Thank you for the numbers. Care to add a row for 2.4.0 candidate ^^? I am not that well set up to produce numbers...

Cool.

sliem, 2018-12-10:
I tested https://github.com/sebi2k1/node-can/commit/618e819f3a59295dc0834f1d937b3852eb78be52 and I got 23-28% (fluctuating) CPU% with 10k fps. A definite improvement from ~40%
Number of `poll` calls is definitely down as shown by this strace:

```
$ timeout 10s strace -qfcp <PID of new socketcan>
% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 90.44    8.473254         193     44009      8333 futex
  2.77    0.259618          29      9049           epoll_pwait
  2.77    0.259541           4     66426      9048 recvfrom
  2.07    0.193569           3     57378           ioctl
  0.68    0.063862           7      9049           write
  0.66    0.062246           7      9048           poll
  0.60    0.056199           6      9049           read
  0.00    0.000217           9        24           brk
  0.00    0.000094          94         1           restart_syscall
------ ----------- ----------- --------- --------- ----------------
```

The best sleep-based code achieved (sleep time 10 ms) reached ~7.5%. The difference likely lies between how often the `async_receiver_ready` is called to handle the frames. We can use the number of `recvfrom` errors to check how many times `async_receiver_ready` is called. **9048** for the new thread locking version, while **985** for the sleepy version (see below for the strace). The additional overhead of switching back and forth is likely account for the performance difference. One way to bridge that performance gap would be to make `async_receiver_ready` less prone to quit after consuming all available frames (`recv` with a socket option `SO_RCVTIMEO` rather than non-blocking?).

```
$ timeout 10s strace -qfcp <PID of sleepy socketcan>
% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 72.99    4.013484       20904       192        31 futex
 19.18    1.054610        1072       984           nanosleep
  3.82    0.210013           4     58247       985 recvfrom
  3.19    0.175674           3     57262           ioctl
  0.50    0.027403          28       985           epoll_pwait
  0.11    0.006177           6       985           read
  0.11    0.006124           6       985           write
  0.08    0.004532           5       985           poll
  0.01    0.000279          11        26           brk
  0.00    0.000097          97         1           restart_syscall
  0.00    0.000038           5         8           mmap
  0.00    0.000031           4         8           munmap
  0.00    0.000023           3         8           mprotect
------ ----------- ----------- --------- --------- ----------------
100.00    5.498485                120676      1016 total
```

sliem, 2018-12-10:
I wrote a quick version with blocking `recv` with 10 ms timeout. Roughly ~14% CPU with +-3% fluctuations.

```
$ timeout 10s strace -qfcp <PID of timeout socketcan>
% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 91.62    5.791686        2495      2321       250 futex
  4.40    0.277967           5     57142           recvfrom
  3.34    0.211139           4     57142           ioctl
  0.42    0.026383          47       566           epoll_pwait
  0.09    0.005838          10       566           write
  0.07    0.004443           8       566           poll
  0.06    0.004010           7       566           read
  0.00    0.000264          14        19           brk
------ ----------- ----------- --------- --------- ----------------
100.00    6.321730                118888       250 total
``` 

Patch:
```patch
--- socketcan/src/rawchannel.cc       2018-12-10 15:03:00.122443351 +0100
+++ timeoutcan/src/rawchannel.cc   2018-12-10 15:26:47.136742578 +0100
@@ -120,6 +120,12 @@
       if (setsockopt(m_SocketFd, SOL_CAN_RAW, CAN_RAW_ERR_FILTER, &err_mask, sizeof(err_mask)) != 0)
         goto on_error;

+      struct timeval tv;
+      tv.tv_sec = 0;
+      tv.tv_usec = 10000;
+      if (setsockopt(m_SocketFd, SOL_SOCKET, SO_RCVTIMEO, (const char*)&tv, sizeof tv) != 0)
+        goto on_error;
+
       memset(&m_SocketAddr, 0, sizeof(m_SocketAddr));
       m_SocketAddr.can_family = PF_CAN;
       m_SocketAddr.can_ifindex = ifr.ifr_ifindex;
@@ -555,7 +561,7 @@

     unsigned int framesProcessed = 0;

-    while (recv(m_SocketFd, &frame, sizeof(struct can_frame), MSG_DONTWAIT) > 0)
+    while (recv(m_SocketFd, &frame, sizeof(struct can_frame), 0) > 0)
     {
       Nan::TryCatch try_catch;

```

juleq, 2018-12-11:
@sliem I gave the patch on top of 2.4.0 a try and I am down to 10-14%. But keep in mind that this is a complete application, so I take my share, too. If I understand the patch correctly, it will have no impact other than sparing the extra rounds of ping pong, right? 

@sebi2k1 Any objections with respect to the patch? Sounds like another good idea :).

juleq, 2018-12-11:
Ok, that explains that I see jitter in my output timer. That js timer samples my signal buffer in 50ms intervals, timestamps and forwards to the output stream. I saw 70ms instead of 50ms there from time to time where it has been +-2ms jitter before.

sliem, 2018-12-12:
Yes, it would block.  If there is no more messages, (perhaps a somewhat rare scenario in real applications?), it would block for 10 ms waiting for a message that does not arrive.

The worst case scenario would be 100 fps, where we could potentially wait (and block) for each message the maximal amount of time (1s).

I tried with an 1ms timeout and got 16 +-3 % CPU usage. 0.1 ms timeout got me very fluctuating number between 13-20%. These might be more reasonable choices with worst case scenarios of 10 ms blocking at 1k fps and 1 ms at 10k fps.

juleq, 2018-12-12:
Ok. I assume the jitter I see is somewhat artificial then: Since I simulate the entire bus with ~16 ECUs/bus, the simulation will output the frames in bursts. The fastest frames are 60ms cycle, the slower ones are multiples thereof. That also means blocking while waiting for the next burst.

I am pretty sure that there is a setting for a startup delay for each ECU (maybe something random...). I‘ll have a look into that when I am back on it.

I am not that familiar with linux, how would a block by recv affect the CPU load counters?

juleq, 2018-12-13:
I have spaced out ECU startup, but that did not have that much impact.

I do see a rough increase of nodejs timer jitter when the patch is applied. To illustrate here is an extract of the logs that I create. It shows the js timer timestamp and the time elapsed since the previous one. Sampling here is 50ms. Conditions are otherwise identical:

**Unpatched**
`14:27:05,573 | 00:00:00,050`
`14:27:05,622 | 00:00:00,049`
`14:27:05,672 | 00:00:00,050`
`14:27:05,723 | 00:00:00,051`
`14:27:05,773 | 00:00:00,050`
`14:27:05,823 | 00:00:00,050`
`14:27:05,873 | 00:00:00,050`
`14:27:05,923 | 00:00:00,050`
`14:27:05,973 | 00:00:00,050`
`14:27:06,024 | 00:00:00,051`

**Patched**
`14:12:35,151 | 00:00:00,050`
`14:12:35,212 | 00:00:00,061`
`14:12:35,262 | 00:00:00,050`
`14:12:35,332 | 00:00:00,070`
`14:12:35,382 | 00:00:00,050`
`14:12:35,452 | 00:00:00,070`
`14:12:35,502 | 00:00:00,050`
`14:12:35,572 | 00:00:00,070`
`14:12:35,622 | 00:00:00,050`
`14:12:35,702 | 00:00:00,080`

I personally would definetly stick to the non-blocking but syncronized solution of the 2.4.0 unptched as it is far less unobtrusive with respect to the other parts of the application. The patch might still be interesting for edge cases that require low load at the expense of latency.

@sebi2k1 Are you planning to release 2.4.0 to npm?

sebi2k1, 2018-12-13:
Yeah, I'm going to release it. Tests went fine here and if you guys didn't encounter other issues I would rather release 2.4.0 w/o the additional blocking patch.

sliem, 2018-12-14:
Didn’t encounter any issues. Keeping it non-blocking is prob the correct call, will likely mean less mystery performance/latency issues for users.

juleq, 2018-12-14:
I‘ve just pulled in the update. It has been a pleasure. Thank you guys.

sliem, 2018-12-14:
Thank you for the new release, I will probably deploy the new version after Christmas.

closed 2018-12-14 by sliem