sebi2k1/node-can #50: Not able to install the lastest version

Bobet21, 2019-01-18:
I downloaded the Github version - unzipped it then executed the following commands
1. cd Downloads/node-can-master
2. npm i 
3. npm run-configure
4. npm run-build
5. npm install socketcan

 and i get the following error

> npm ERR! code ENOSELF
npm ERR! Refusing to install package with name "socketcan" under a package
npm ERR! also called "socketcan". Did you name your project the same
npm ERR! as the dependency you're installing?
npm ERR! 
npm ERR! For more information, see:
npm ERR!     <https://docs.npmjs.com/cli/install#limitations-of-npms-install-algorithm>

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-01-18T21_00_39_714Z-debug.log

juleq, 2019-01-18:
I think you are trying to install into the module source directory. Try changing to your project directory and do the install there. If you install from source folder, you need to specify the source folder then, e.g. npm install ~/Downloads/node-can-master/socketcan or something...

Bobet21, 2019-01-20:
@sebi2k1 , i missed the point sorry, i thought it was 1. then 2....without thinking much..will the node be added automatically to the palette ? The original reason i went through gitbhub is because adding those nodes through the palette was not working... now after a simple npm install and a reboot i can still not see them..

> make : on quitte le répertoire « /home/pi/node_modules/socketcan/build »
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
+ socketcan@2.4.0
updated 1 package and audited 11 packages in 18.147s
found 0 vulnerabilities

sebi2k1, 2019-01-20:
Not sure if I get your question right. If you want to install socketcan globally just do:
`npm install -g socketcan`

After that, socketcan is available by just during "require("socketcan")" as indicated by the example.

Is it that want you want to achieve?

Bobet21, 2019-01-20:
well, i realize i may just have misplaced my request - I am using the Node-red graphical tool - before i installed the node-red-contrib-canbus through the palette and get access to 2 node can-send and can-dump. 

Yesterday the install through the palette failed so i tried using the link provided there
[https://flows.nodered.org/node/node-red-contrib-canbus]- leading to your rep..

once done the install using npm i cannot see those 2 nodes in the palette.Installing globally  failed (even using sudo) because of permissions

> gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/10.15.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/socketcan/build'
gyp ERR! System Linux 4.14.79-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/socketcan
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! socketcan@2.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the socketcan@2.4.0 install script.

Bobet21, 2019-01-22:
[image]
[image]

See attached pictures - my node candump is not recognized...

sebi2k1, 2019-01-24:
Seems to be a node-gyp issues in your environment. I did not tried node-red yet so I cannot really help.

closed 2019-01-24 by sebi2k1

Bobet21, 2019-02-01:
Some reccomendation i went across are to use nvm instead of npm for the install... any advice from your side?