sebi2k1/node-can #51: Remote request kills network?

RobertSmart, 2019-02-11:
Hi,

I'm using a bog standard MCP2515 based can board to communicate with another device. It seems to get all the data just fine until I get message like this though.

can0  0DFFFFFF   [8]  remote request

That seems to kill the network and I need to down/up can0 to get it working again until the next remote request. Is there a way to stop this behaviour?

Thanks,

Rob

sebi2k1, 2019-02-11:
Mh, that's odd. It is an extended remote transfer request. You have a candump trace with mask and link statistics?

RobertSmart, 2019-02-12:
Hi,

I ran candump with options -d -e -x set. as you can see it just stops after the remote request message. I have to down/up the network to get it candump running again.

  can0  RX - -       35E   [8]  50 59 4C 4F 4E 20 20 20
  can0  RX - -       305   [8]  00 00 00 00 00 00 00 00
  can0  RX - -       305   [8]  00 00 00 00 00 00 00 00
  can0  RX - -       359   [7]  00 00 00 00 02 50 4E
  can0  RX - -       351   [6]  14 02 C8 00 F4 01
  can0  RX - -       355   [4]  61 00 64 00
  can0  RX - -       356   [6]  2F 14 BF 00 B2 00
  can0  RX - -       35C   [2]  C0 00
  can0  RX - -       35E   [8]  50 59 4C 4F 4E 20 20 20
  can0  RX - -       359   [7]  00 00 00 00 02 50 4E
  can0  RX - -       351   [6]  14 02 C8 00 F4 01
  can0  RX - -       355   [4]  61 00 64 00
  can0  RX - -       356   [6]  2B 14 BE 00 B2 00
  can0  RX - -       35C   [2]  C0 00
  can0  RX - -       35E   [8]  50 59 4C 4F 4E 20 20 20
  can0  RX - -       305   [8]  00 00 00 00 00 00 00 00
  can0  RX - -       305   [8]  00 00 00 00 00 00 00 00
  can0  RX - -       359   [7]  00 00 00 00 02 50 4E
  can0  RX - -       351   [6]  14 02 C8 00 F4 01
  can0  RX - -       355   [4]  61 00 64 00
  can0  RX - -       356   [6]  2B 14 BE 00 B2 00
  can0  RX - -       35C   [2]  C0 00
  can0  RX - -       35E   [8]  50 59 4C 4F 4E 20 20 20
  can0  RX - -  0DFFFFFF   [8]  remote request

sebi2k1, 2019-09-21:
Likely a controller issue and not node-can related.

closed 2019-09-21 by sebi2k1