sebi2k1/node-can #52: node-can issues error in building on (node v11 and node-gyp v3)

Meghadoot, 2019-02-13:
I'm new to node environment. I wanted to have a can node (for transmission and reception of simple can frames) in node-red which is on raspberry pi-3B.  
I tried to install a node from palette from node-red, as per mentioned in:
(https://flows.nodered.org/node/node-red-contrib-canbus)
But installation failed. 
So i tried installation with terminal command and faces build errors.

(Note: I am working on node v11.9.0 and npm 6.5.0.)  

I guess it is related to version 8 node. Here is the terminal window details:
```````````````````````````````````````````````````````````````````````````````````````````````

pi@raspberrypi:~/.node-red $ npm install node-red-contrib-canbus

> socketcan@2.1.3 install /home/pi/.node-red/node_modules/node-red-contrib-canbus/node_modules/socketcan
> node-gyp rebuild

make: Entering directory '/home/pi/.node-red/node_modules/node-red-contrib-canbus/node_modules/socketcan/build'
  CXX(target) Release/obj.target/can/src/rawchannel.o
../src/rawchannel.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE RawChannel::New(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rawchannel.cc:166:45: warning: ‘v8::Local<v8::String> v8::Value::ToString() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10248): Use maybe version [-Wdeprecated-declarations]
     Nan::Utf8String ascii(info[0]->ToString());
                                             ^
../src/rawchannel.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE RawChannel::AddListener(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rawchannel.cc:200:50: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10254): Use maybe version [-Wdeprecated-declarations]
         listener->handle.Reset(info[2]->ToObject());
                                                  ^
../src/rawchannel.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE RawChannel::Send(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rawchannel.cc:260:52: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10254): Use maybe version [-Wdeprecated-declarations]
     v8::Local<v8::Object> obj =  info[0]->ToObject();
                                                    ^
../src/rawchannel.cc:263:53: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:2571): Use maybe version [-Wdeprecated-declarations]
     frame.can_id = obj->Get(id_symbol)->Uint32Value();
                                                     ^
../src/rawchannel.cc:276:60: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10254): Use maybe version [-Wdeprecated-declarations]
     frame.can_dlc = node::Buffer::Length(dataArg->ToObject());
                                                            ^
../src/rawchannel.cc:277:61: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10254): Use maybe version [-Wdeprecated-declarations]
     memcpy(frame.data, node::Buffer::Data(dataArg->ToObject()), frame.can_dlc);
                                                             ^
../src/rawchannel.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE RawChannel::SetRxFilters(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rawchannel.cc:322:53: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10254): Use maybe version [-Wdeprecated-declarations]
         if (ObjectToFilter(list->Get(idx)->ToObject(), &rfilter[numfilter]))
                                                     ^
../src/rawchannel.cc:332:44: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10254): Use maybe version [-Wdeprecated-declarations]
       if (ObjectToFilter(info[0]->ToObject(), &rfilter[numfilter]))
                                            ^
../src/rawchannel.cc: In static member function ‘static bool RawChannel::ObjectToFilter(v8::Handle<v8::Object>, can_filter*)’:
../src/rawchannel.cc:414:39: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:2571): Use maybe version [-Wdeprecated-declarations]
     rfilter->can_id = id->Uint32Value();
                                       ^
../src/rawchannel.cc:415:43: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:2571): Use maybe version [-Wdeprecated-declarations]
     rfilter->can_mask = mask->Uint32Value();
                                           ^
../src/rawchannel.cc: In member function ‘void RawChannel::async_receiver_ready(int)’:
../src/rawchannel.cc:486:32: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../../../nan/nan.h:1674) [-Wdeprecated-declarations]
           callback.Call(1, argv);
                                ^
../src/rawchannel.cc:488:60: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../../../nan/nan.h:1652) [-Wdeprecated-declarations]
           callback.Call(Nan::New(listener->handle), 1, argv);
                                                            ^
  SOLINK_MODULE(target) Release/obj.target/can.node
  COPY Release/can.node
  CXX(target) Release/obj.target/can_signals/src/signals.o
../src/signals.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE DecodeSignal(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/signals.cc:103:46: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/11.9.0/include/node/v8.h:10254): Use maybe version [-Wdeprecated-declarations]
     Local<Object> jsData = info[0]->ToObject();
                                              ^
../src/signals.cc:111:35: error: no matching function for call to ‘v8::Value::ToUint32()’
     offset    = info[1]->ToUint32()->Uint32Value();
                                   ^
[... 102 lines cut ...]
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:197:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:254:12)
gyp ERR! System Linux 4.9.35-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/.node-red/node_modules/node-red-contrib-canbus/node_modules/socketcan
gyp ERR! node -v v11.9.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! socketcan@2.1.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the socketcan@2.1.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-02-13T01_16_19_121Z-debug.log
``````````````````````````````````````````````````````````````````````````````````````````````````````

looking forward to hear from you.
thanks in advance.

sebi2k1, 2019-02-13:
Was already fixed with 2.4.0. node-red is depending on a old version. Please check with the node-red package maintainer.

closed 2019-02-13 by sebi2k1

Bobet21, 2019-02-13:
see the following node-red issue we would need to raise this to the node owner
[https://discourse.nodered.org/t/node-red-contrib-canbus-install-failed/7160/2]

Meghadoot, 2019-02-13:
@sebi2k1 @Bobet21 
Thanks for replys. 

so can you please suggest, is there any way or tutorial to work with socketcan node in node-red (to send and receive can standard frames on raspberry-pi)?
instead of downgrading node version for the sake of working with canbus node which works for node v8.

d21d3q, 2019-02-22:
Just check out example flow from [here](https://flows.nodered.org/node/node-red-contrib-canbus). If `can-utils` are working properly on your machine, then flow should work properly. If not, follow [this](https://www.raspberrypi.org/forums/viewtopic.php?t=141052) to setup can on rpi.