sebi2k1/node-can #54: Keeping track of features to add.

juleq, 2019-03-03:
Hello @sebi2k1,

I was wondering how you would like to keep track of features that wait for implementation. Should we add a new issue for each observation that might get addressed in the future?

What I have noticed so far:

-Multiplexing support is limited in that only one multiplexor can be present in a message
-Multiplexing support is limited in that the multiplexor signal is expected to be of length 8 at position 0
-Maximum signal length supported appears to be 32 bit
-No floating point support

I know a nice dbc file to test these enhancements when they have been implemented. The lab power supply manufacturer ElektroAutomatik uses e.g. 16 bit big endian multiplexors for 32bit floating point nominal values (e.g. device nominal current, etc.). They provide the dbc files for their units free of charge and they can be converted to dbc e.g. by the Python package cantools.

I cannot contribute right now but I thought I make a note for future me (and perhaps you :).

sebi2k1, 2019-03-03:
Would prefer to create issues to track feature requests.

juleq, 2019-03-03:
Ok. I have opened issues with hopefully meaningful titles.

Thanks again for your project. I am a huge fan :).

closed 2019-03-03 by juleq