sebi2k1/node-can #55: Multiplexing limitation: No extended multiplexing support.

juleq, 2019-03-03:
The current implementation does not support extended multiplexing.

> Extended multiplexing allows defining several multiplexors in a message and a signal can be multiplexed for more than one multiplexor value. A multiplexed signal can also serve as multiplexor signal for other multiplexed signals, meaning that hierarchical multiplexing can be modeled. 

-> Extended multiplexing would be a great enhancement but likely requires a little remodeling of the current object structure. That could require breaking changes.

The quote above is taken from the Vector document below, but the kcd file format also supports it:
[Extended Multiplexing](https://assets.vector.com/cms/content/know-how/_application-notes/AN-ION-1-0521_Extended_Multiplexing_in_DBC_Databases.pdf)

sebi2k1, 2019-09-21:
Multiple muxing groups are already supported. You got an example for KCD referencing the same signal in multiple groups and using a signal as a muxer?

juleq, 2019-09-25:
Hi. I think extended multiplexing is not that important since the standard version of CANdb++ is not supporting it either. KCD is very flexible and I would not advocat that supporting that kind of flexibility is that important either. But understanding multiplexors to be regular signals with various type/length/positioning is quite common and supported by CANdb++. Should I upload such a db as dbc/kcd for you?