sebi2k1/node-can #59: Signal type limitation: No support for float/single and double.

juleq, 2019-03-03:
The current implementation does not support signals of floating point types single or double.

sebi2k1, 2019-09-14:
It is quite uncommon though in CAN to transfer raw float values and I believe it is not standardise at all. We can follow the CANopen specification for encoding floats in a PDO, but I do not have access to the specification anymore. :-/

Any ideas?

juleq, 2019-09-14:
Hi. Yes, it‘s uncommon and presently I shake my head when I stumble over this. As an example: ElektroAutomatik has some kind of multiple interface protocol that also uses floats for their PSUs. You can find their dbcs online.

Implementink is no trouble: IEEE-754 is used. You can just cast your float/double into a byte array. If dbc endianness deviates from machine, you would have to do that dance, too.

I assume that floats might get more prevalent to buses in the industry in the future because the usual suspects in ECUs these days are getting more and more of armhf type. Some even do double in HW and if you fall back to SW, the performace hit is not as bad as it was considering the increased overall performance.