sebi2k1/node-can #63: Support Node >8

team-p5x, 2019-08-14:
Hi,

i have problems installing node-can with npm when having a Node Version greater than 8.
Is there a trick or am i doing something wrong? 
I tried 'npm install node-can' command on a linux machine and got errors concerning to 'make'.

My error pretty much looked like in
#52

sebi2k1, 2019-08-15:
The package was renamed to socketcan a couple of months ago.

team-p5x, 2019-08-15:
Excuse me, of course I typed "npm install socketcan".

team-p5x, 2019-08-16:
Can you please reopen the issue?

kais0r, 2019-09-02:
Hi. 

There is still a Problem using Node >8 with the Raspberry Pi.

had problems with v10.16.3
downgraded to v8.16.0 and it was ok.

Wanted to use: node-red-contrib-canbus
which needs socketcan to work.

So... it node v8 is a option, you an use it as workarround...:
`node -v` to show actual version
`sudo npm install -g n`
`sudo n 8.16.0`
`node -v`

juleq, 2019-09-02:
Hi.

I am cross compiling for arm against various versions of node 10 without a single issue. I remember someone using node-red-contrib-canbus having an issue. I think the problem there was depending on an outdated version of node-can. Which version of node-can are you trying to build?

kais0r, 2019-09-02:
Hi,

node-can (socketcan) version is:     socketcan@^2.4.0

juleq, 2019-09-02:
Hi,

does it error out in the same way when you try to install it via npm as a standalone package?

juleq, 2019-09-02:
Just as another datapoint: Have updated to 10.16.3 -> also builds fine (npm standalone install).

kais0r, 2019-09-03:
I think i got the problem. I installed node-red-contrib-canbus via the node-red palette manager.

in the actual `node-red-contrib-canbus/package.json` there is the part :
`"dependencies": {
    "socketcan": "2.1.x",
    "random-js": "1.0.x"
  },`

So.. what it then does: It installs socketcan v2.1.3 as another "submodule" 
`~/.node-red/node_modules/node-red-contrib-canbus/node_modules/socketcan`

The problem is the outdated package.json from `node-red-contrib-canbus`

Sorry i did not see it before...

sebi2k1, 2019-09-07:
Closed as issue was in local configuration.

closed 2019-09-07 by sebi2k1