sebi2k1/node-can #64: Change 'associative array' to Object or other Datatype

team-p5x, 2019-08-20:
In socketcan.js you are using a associative array for the signals. Why don't you use it as a normal array or why don't you use a Object?

`/**
         * Named array of signals within this message. Accessible via index and name.
         * @attribute {Signal} signals
         * @final
         */
	this.signals = [];

	for (i in desc['signals']) {
		var s = desc['signals'][i];
		if (this.signals[s.name] && this.signals[s.name].muxGroup) {
			this.signals[s.name].muxGroup.push(s.mux);
		} else {
			this.signals[s.name] = new Signal(s);
		}
	}`

sebi2k1, 2019-09-07:
Sorry for the late reply, just back form vacation. I am not sure what you are actually proposing? What is the disadvantage from your side using an associative array?

sebi2k1, 2019-12-31:
Feel free to make a pull request, may be I understand better then what you are trying to enhance.

closed 2019-12-31 by sebi2k1