sebi2k1/node-can #74: Problems installing on arch

ptiza-v-nebe, 2019-12-01:
Hi, I run arch linux with latest upgrade on raspi 3.

$ node -v
v13.2.0

$npm -v
6.12.1

Trying to install node-can:
$ npm install socketcan

and failing with this error:
...
../src/rawchannel.cc: In static member function 'static void RawChannel::Init(v8::Local<v8::Object>)':
../src/rawchannel.cc:101:97: error: no matching function for call to 'v8::Object::Set(v8::Local<v8::String>, v8::Local<v8::Function>)'
     exports->Set(Nan::New("RawChannel").ToLocalChecked(), Nan::GetFunction(tpl).ToLocalChecked());
..

and similiar

..

gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:270:12)
gyp ERR! System Linux 4.19.86-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/projects/node_modules/socketcan
gyp ERR! node -v v13.2.0
gyp ERR! node-gyp -v v6.0.1
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/projects/package.json'
npm WARN projects No description
npm WARN projects No repository field.
npm WARN projects No README data
npm WARN projects No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! socketcan@2.5.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the socketcan@2.5.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-12-01T16_39_48_827Z-debug.log

same behavior with installing from source
probably nan api has changed heavily

ptiza-v-nebe, 2019-12-01:
Ok, the story gets more interesting..

For some reason by trying to install from github we get errors all over the place
where formerly deprecated functions cannot be used:
git clone 

If we try to install from npm repository with
$ git clone https://github.com/sebi2k1/node-can
$ cd node-can
$ npm i

gives error of type:
../src/rawchannel.cc: In static member function 'static void RawChannel::Init(v8::Localv8::Object)':
../src/rawchannel.cc:101:97: error: no matching function for call to 'v8::Object::Set(v8::Localv8::String, v8::Localv8::Function)'
exports->Set(Nan::New("RawChannel").ToLocalChecked(), Nan::GetFunction(tpl).ToLocalChecked());

If we try to install from npm repository:
$ npm install socketcan

We get warnings of type:
../src/rawchannel.cc: In static member function 'static void RawChannel::Init(v8::Local<v8::Object>)':
../src/rawchannel.cc:102:97: warning: 'bool v8::Object::Set(v8::Local<v8::Value>, v8::Local<v8::Value>)' is deprecated: Use maybe version [-Wdeprecated-declarations]
     exports->Set(Nan::New("RawChannel").ToLocalChecked(), Nan::GetFunction(tpl).ToLocalChecked());

And 1 error:
../src/rawchannel.cc: In member function 'void RawChannel::async_receiver_ready(int)':
../src/rawchannel.cc:582:38: error: 'SIOCGSTAMP' was not declared in this scope
         if (likely(ioctl(m_SocketFd, SIOCGSTAMP, &tv) >= 0))

which is solved by including:
#include <linux/sockios.h>
to rawchannel.cc

version of nodejs v12.13.1 and npm 6.12.1

ptiza-v-nebe, 2019-12-01:
Correction on the first statement. socketcan 2.6.0 from github repo compiles with nan 2.14.0

$ git clone https://github.com/sebi2k1/node-can
$ cd node-can
$ npm i

again compiles with very much deprecated warnings

ptiza-v-nebe, 2019-12-01:
Problem persists only on arm architecture.
Normal x86 computer install from npm repo out of box

sebi2k1, 2019-12-04:
Published 2.6.0 on the weekend which includes the sockios.h fixes. On raspberry pi it fails because of nodejs v13?

sebi2k1, 2019-12-31:
No feedback for 27d.

closed 2019-12-31 by sebi2k1