sebi2k1/node-can #75: Add method to set bitrate

dawn-minion, 2019-12-09:
Would it be possible to add a method to set the bitrate/baudrate? Module works great for me, but I need to switch the bitrate at runtime based upon input from one of the CAN devices on our network.

Right now I just have to shell out to the `ip` command which works, but it'd be really nice if there was something similar to libsocketcan's `set_can_bitrate` method to set bitrate for a specific interface: https://lalten.github.io/libsocketcan/Documentation/html/group__extern.html#gaf69a4a0d7424e5a61d60991295f50559

sebi2k1, 2019-12-31:
It requires integration of libsocketcan which is not used right now. But I like the idea as it also offers restarting the device and so on.

Unfortunately I do not have actual hardware available yet to test if the bitrate change actually.

Hyphen90, 2021-01-11:
Hi, are there any plans to support libsocketcan in the near future? It would be perfect, if I could set the bitrate and message filter directly  from your code.

sebi2k1, 2021-01-11:
Yes, indeed. I didn’t checked yet if the lib is already in Debian/Ubuntu tepid, would it make easier for the test builds.