sebi2k1/node-can #77: Node process stalling on specific target

juleq, 2020-01-06:
Hi,

I have one target system (of three) where I get the following behaviour: After ~1500 seconds, the node process will get stuck when the module using node-can is activated. Not a single timer of the node process will fire anymore. the process just sits there. It also does not get scheduled anymore (that is by judging `cat /proc/<PID>/sched` and `status`, these will show frozen values). 

`cat /proc/<PID>/syscall` will yield the following:
```
0 0x20 0x7ea7c000 0x10 0x0 0x7ea7c010 0x7ea7c048 0x7ea7bfe0 0x76d8d7d4
```
I would assume that this means that the last syscall was 0. That seems to be a syscall internal to the kernel, I am not sure what to make of that... Unfortunaltely I have no easy way to get a stack trace.

To find out which syscall it is stuck in, I tired `ps -p <PID> -ocmd,stat,wchan`, which got me:
```
CMD               STAT WCHAN
node App.js       Sl   sock_alloc_send_pskb
```

Do you have any suggestions what I could try to pinpoint this issue?

juleq, 2020-01-06:
Addendum: I have two CAN channels activated but no busses connected when the problem described above happens. When I connect the busses to each other (one channel sends a few frames every minute), the problem won´t show.

sebi2k1, 2020-01-06:
Sounds like the RawChannel::Send call suddenly blocks due to full Tx buffer on controller side as the transmit is potentially blocking.

What is the use case to send something on channel w/o connected bus?

juleq, 2020-01-06:
The application requests data periodically. It does not care if a node is present. From time to time it happens that the bus gets disconnected (e.g. due to maintenance). It would be good to have tx queue full as an error that I have the chance to recover from. With the current behavior I (the whole node process) am dead in the water 25 mins after I have been unplugged :).

I have a test running now to confirm your assumption. I do not send, everything else is the same. If it works, you are correct.

sebi2k1, 2020-01-06:
We could introduce a poll(POLLOUT) right before sending to check if the Tx buffer is available before send() to avoid block, or introduce an additional parameter to perform a non-blocking send and return error if sending failed. Would that help?

juleq, 2020-01-06:
When sending non-blocking, will send return immediately if it could not obtain the tx buffer? 

From the users perspective, I think that immediate feedback is helpful. So call send and check whats returned if that concerns. In my case I just want to fire and forget.

juleq, 2020-01-06:
So maybe:
- check tx buffer availability
- return -1 if buffer is not available (that is what the send of socket.h also does in case of an error)
- send blocking otherwise

What do you think?

sebi2k1, 2020-01-06:
Yea makes sense, but need to provide an API extension to not break the behaviour for others. Also need to find a way to reproduce that behaviour ideally using vcan for integration tests.

juleq, 2020-01-06:
But the behaviour remains almost the same. And it will prevent the blocking (that could not be recovered from). What behaviour will that break for existing users?

sebi2k1, 2020-01-06:
I suppose you can make it unblock by just connecting the bus in that very moment. Depending on the application and structure a full Tx buffer may not be uncommon (considering a low baud rate bus and high frequency sender). These application may rely on the current behaviour and accepts a short blocking before Tx buffer gets available.

juleq, 2020-01-06:
I see. Another approach would be an option to use non-blocking sends when creating the channel. Or a parameter to the send function.

I think I like the idea of an options object passed to the createRawChannel. Perhaps:
```
createRawChannelWithOptions ("can0", {
  timestamps: true,
  protocol: 0,
  nonBlockingSend: true
});
```

sebi2k1, 2020-01-07:
Give it a try f34d1a67796c2c3db40a927dd42d0523c17b588a

juleq, 2020-01-07:
Wow, that was fast :). Thanks.

Am I missing something? I do not see the flag getting applied. I googled how thats done for networking sockets:
>Call the fcntl() API to retrieve the socket descriptor's current flag settings into a local variable.

>In our local variable, set the O_NONBLOCK (non-blocking) flag on. (being careful, of course, not to tamper with the other flags)

>Call the fcntl() API to set the flags for the descriptor to the value in our local variable.

Shouldn’t that be the same for CAN sockets? Or was the approach to just do the send non-blocking and I missed how you did that?

juleq, 2020-01-07:
Ahh, I got it by looking at sends signature: I think the 0 should have been a flags. I will try that.

sebi2k1, 2020-01-07:
Yeah, I use the MSG_DONTWAIT in RawChannel::Send to enable non-blocking access.

juleq, 2020-01-08:
Fantastic. That does the trick. I get -1 when the tx buffer runs full but I can continue trying to send. As soon as CAN bus is connected, this will recover and send frames again.

I can also confirm that the blocking version will continue once the CAN bus is available again. But the whole node process will be blocked. In my case that also means: no more servicing of the watchdog and consecutively an unexpected reboot.

I have tested what has been described so far. I have not executed the tests that you provided. Should I do that?

Regarding the blocking version: The node people strongly advise against blocking the event loop. They even cite synchronous processing of long JSON inputs as an example. In this socketcan case we can be blocked from the outside with no control left whatsoever. That is not what your average node developer is prepared for and it sure caught me by surprise :). I remember a blocking discussion on the read thread sync. I would say the situation is different in that it could not stall indefinetly, right?

That being said maybe pointing that out in the docs and referencing this new shiny option is the most strightforward approach to prevent others from hitting this block :).

Please take this big fat **thank you** for your effort and support.

You mention from time to time that you have no HW available to test. Have you ever considered a Raspberry with a CAN hat, e.g. like PiCan2 DUO? I have used that sucessfully with 2 CAN channels. It uses SPI to connect two external CAN controllers since the RPI still has no CAN IP in its SoC.

sebi2k1, 2020-01-08:
I agree that blocking behaviour is not expected - me too apparently. It will unblock only if the tx buffer gets available at some point. But still if it happens you are screwed. ;-)

closed 2020-01-08 by sebi2k1

juleq, 2020-02-18:
Any plans on pushing this to npm ^^?

sebi2k1, 2020-02-20:
Thanks for the reminder. Just published 2.7.0.