sebi2k1/node-can #80: Signal array is being passed a string instead of an index

soupyt, 2020-03-12:
In the Message() function in socketcan.js, the empty array `this.signals` is addressed with a string (like an object, not an index:

        this.signals = {};

        for (i in desc['signals']) {
                var s = desc['signals'][i];
                console.log(JSON.stringify(s, null, '  '));
                if (this.signals[s.name] && this.signals[s.name].muxGroup) {
                        this.signals[s.name].muxGroup.push(s.mux);
                } else {
                        this.signals[s.name] = new Signal(s);
                }
        }

This causes the error:
```
socketcan.js:252
			this.signals[s.name] = new Signal(s);
			                     ^

RangeError: Invalid array length
```
Should `this.signals` be initialized as an object instead?

sebi2k1, 2020-03-13:
How does the database definition look like? We are using the signals as a dictionary so the code should be right basically.

soupyt, 2020-03-17:
Sorry for the delay, a lot going on these days . . .

Do you need the entire database definition?

I started with a DBC file from https://www.csselectronics.com, which states:

> This OBD2 DBC demo contains most of the standardized OBD2 PIDs from Mode 01.
It does not contain proprietary PIDs, though you can freely expand the DBC as you see fit.

> Note that the OBD2 DBC uses complex multiplexing to achieve conversion of the data.
This follows the DBC standard, but few tools support it.

Which I then converted to a kcd file using canmatrix. This yielded a XML document of the form:
`<NetworkDefinition xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://kayak.2codeornot2code.org/1.0" xsi:schemaLocation="Definition.xsd">
  <Document name="Some Document Name">some text</Document>
  <Bus name="OBD2">
    <Message id="0x7E8" name="OBD2" length="8">
      <Notes>OBD2 DBC developed by canmatrix &amp; CSS Electronics - for questions contact us at www.csselectronics.com</Notes>
      <Multiplex name="service" offset="8" length="4">
        <MuxGroup count="0">
          <Signal name="S1_PID_00_PIDsSupported_01_20" offset="24" length="32" endianess="big">
            <Notes>PIDs supported [01 - 20] - Bit encoded [A7..D0] == [PID $01..PID $20]. See OBD-II PIDs on Wikipedia</Notes>
            <Value max="4294967295"/>
          </Signal>
        </MuxGroup>`

By inserting some debug in your code, I was able to determine this was parsed (correctly IMHO) into a JS object, with an equivalent structure. The problem occurs when that object is assigned to the array `this.signals`.

Out of curiosity, I changed line 244 of socketcan.js to `this.signals = {};` and it executes without error. I have not had the opportunity to test the code in a live system however.

sebi2k1, 2020-04-19:
@soupyt Which node version are you using actually? I tested your bus definition on all supported NodeJS version (8, 10, 12 and 13) and it doesn't through the error as you report.

See https://travis-ci.com/github/sebi2k1/node-can/builds/160913318 for build logs of the test run.

closed 2020-04-19 by sebi2k1