sebi2k1/node-can #86: Length not apply from kcd file when sending signals

TIB28, 2020-05-15:
Hello I'm working on sending message using kcd file but i cannot send message with a length of 16.

Here is an extrait of my kcd file: 
```
<Message id="0x037" name="blueLight">
        <Signal name="blueLight" offset="24" length="8"/>
	<Signal name="blueLightSet" offset="0" length="16"/>
</Message>
```

and my code:
```
dbCan.messages["blueLight"].signals["blueLightSet"].update(0x01);
dbCan.send('blueLight');
```

So how can i send with a length of 16 ??? Thanks

sebi2k1, 2020-05-15:
Interesting, we actually have a unit test exactly for that use case. How does the message look like on CAN? What is the result?

juleq, 2020-05-15:
First: Naming signal and message the same probably invites for confusion.

Second: I see one message with two signals. The first defined signal starts at bit 24 and is 8 bits long -> I would assume the message length to be at least 32 bits. That is a length of 4 bytes. The length in bytes is the usual way to specify message length in can bus and is called DLC.

If you do not specify a „length“ for the message, it will be „auto“. So it will be the smallest DLC that can fit all specified signals.

Maybe providing a little more information about what you are trying to do would help to get further assistance.

Best regards.

juleq, 2020-05-15:
Ahh. Maybe just remove your first signal in the kcd. The resulting message will have a DLC of 2, which is the 16 bits you are looking for... Is that what you were trying?

TIB28, 2020-05-15:
@sebi2k1 the message on can is 01 00 00 00 so a DLC of 2.

TIB28, 2020-05-15:
@juleq when I remove the first signal in my kcd I can send data with a DLC of 2 but i need to listen to another tram with a DLC of 4 on the same id

juleq, 2020-05-15:
Uff. That might be dangerous. 

1.) Bus arbitration is done using the id. That means you can set the message priority by selecting a low id. The sending node that transmits the lowest id will win the bus if multiple nodes try to send at the same time. It is unusual to have two nodes transmitting the same id.

2.) Using the same id for data with a different meaning can produce unexpected behavior: What if you hit the emergency break bit but you meant the message to enable the wiper?

I would suggest being very careful if you are connecting to a complex system. Maybe use the listen only mode and make yourself familiar with it. Get review and permission before writing to a bus that has nodes that are not under your sole control.

TIB28, 2020-05-16:
@juleq What do you mean by emergency break bits ? And enable the wiper ?

TIB28, 2020-05-16:
Is there a way to do multiplexing but with the length?

sebi2k1, 2020-05-16:
Messages are streamed bitwise, once the receiver doesn't see its own signal it stops sending its own message. As the dominant bit wins (CAN_L pulled to GND if I recall correctly) the first low signal wins, which can even be the length byte right after the ID.

But again @juleq is right, using same ID by multiple nodes is quite uncommon and potentially dangerous.

Can you give more details about the application?

juleq, 2020-05-17:
> @juleq What do you mean by emergency break bits ? And enable the wiper ?

That is just an example: When you connect e.g. to some kind of vehicle that hosts a bunch of nodes you might get yourself into trouble when writing to the bus while not fully understanding the consequences: You might e.g. set an emergency break request with the breaking system controller when you just intended to enable the windshield wipers. That’s purely hypothetical to provoke some awareness for the damage that you could do writing to a bus that is not fully understood. If you are just tinkering in a lab: Break it, fix it, repeat :).

TIB28, 2020-05-18:
The tram for enabling the blueLight looks like that when it's off:

[image]

When i Send a Tram with 0100 The tram looks like that which is when the blue light is enabled

[image]

TIB28, 2020-05-18:
Ok i finally just make another kcd file just for sending trams with a DLC of 2.
And the other file is just used to receive trams with a DLC of 4.

closed 2020-05-19 by TIB28