sebi2k1/node-can #87: Multiplex: Receiving value from 2 different signals on the same signal

TIB28, 2020-05-18:
Hey
I need to get the value from my battery but when i get value from one it send me the right info but when i get the same info from the other battery i get it from the same signal name of the first battery

Here is my output code: 

```
Signal AmperageB1 has been sent to Redis with value false.
Signal AmperageB1 has been sent to Redis with value -2000.
Signal AmperageB1 has been sent to Redis with value false.
Signal AmperageB1 has been sent to Redis with value -2000.
Signal AmperageB1 has been sent to Redis with value false.
```
And my kcd:

```
<Message id="0x206" name="IBS">
				<Multiplex name="Mux" offset="0" length="16">
					<MuxGroup count="01">
						<Signal name="AmperageB1" offset="16" length="16" endianess="big">
							<Value slope="0.1" intercept="-2000" unit="A"/>
						</Signal>
						<Signal name="UID_IBS_B1" offset="32" length="16" endianess="big"/>
						<Signal name="PercentageB1" offset="48" length="8">
						<Value slope="1" unit="%"/>
						</Signal>
						<Signal name="SOH_B1" offset="56" length="8"/>
					</MuxGroup>
					<MuxGroup count="257">
						<Signal name="AmperageB2" offset="16" length="16" endianess="big">
							<Value slope="0.1" intercept="-2000" unit="A"/>
						</Signal>
						<Signal name="UID_IBS_B2" offset="32" length="8" endianess="big"/>
						<Signal name="PercentageB2" offset="48" length="8">
							<Value slope="1" unit="%"/>
						</Signal>
						<Signal name="SOH_B2" offset="56" length="8"/>
					</MuxGroup>
					<MuxGroup count="33">
						<Signal name="Capa_B1" offset="16" length="8"/>
						<Signal name="Type B1" offset="40" length="8">
							<LabelSet>
								<Label name="Liquide" value="0"/>
								<Label name="AGM" value="1"/>
								<Label name="Gel EFB" value="2"/>
						</LabelSet>
						</Signal>
					</MuxGroup>
					<MuxGroup count="289">
						<Signal name="Capa_B2" offset="16" length="8"/>
						<Signal name="Type B2" offset="40" length="8">
							<LabelSet>
								<Label name="Liquide" value="0"/>
								<Label name="AGM" value="1"/>
								<Label name="Gel EFB" value="2"/>
						</LabelSet>
						</Signal>
					</MuxGroup>
					<MuxGroup count="17">
						<Signal name="TemperatureB1" offset="16" length="8">
							<Value slope="1" intercept="-40" unit="°C"/>
						</Signal>
					</MuxGroup>
					<MuxGroup count="273">
						<Signal name="TemperatureB2" offset="16" length="8">
							<Value slope="1" intercept="-40" unit="°C"/>
						</Signal>
					</MuxGroup>
				</Multiplex>
			<Signal name="Info IBS" offset="0" length="16"/>
		</Message>
```

juleq, 2020-05-18:
Are you communicating with (1) two BMS (one for each battery) or with (2) one BMS that controls both of the batteries?

If (1): The BMSes should not use the same message id.

If (2): Your kcd suggests that the BMS uses just one id for all information and has a multiplex field longer than one byte to walk through the data on that id. While that is perfectly fine, having a multiplex field longer than one byte and rather sparsely distributed multiplex values (e.g. 1...257) seems suspicious. Whats all the multiplex values in between? You should verify that you have correctly understood the CAN bus definition of the BMS.

TIB28, 2020-05-18:
Well i have one IBS for each battery(so 2) mapped to a single id on can

juleq, 2020-05-18:
So one IBS is one independent CAN node with its own CAN controller. That is a problem.

Usually using multiple similar systems on one bus is done by either:
- Setting the base id of each system to a different id. Further ids the system uses are derived from that by thesystem itself, e.g. BaseID = 0x100; RequestID = BaseID + 0x0; ResponseID = BaseID + 0x1, etc. Your IBS would need to have a parameter like that for you to set.
- Setting the base id by electrical IO. Similar, but instead of a parameter, a voltage level at an IBS input modifies the base id.
- Having two IBS firmwares, each of which preconfigured with its own base id.

juleq, 2020-05-18:
Multiplexing in CAN bus allows ONE node to multiplex a lot of signals and use just one message id for transmission. A multiplex field denotes the set of signals that have been packed into one specific message.

TIB28, 2020-05-18:
I cannot change my IBSs to a separate ids for each it's not under my controls there is no way.
I can only receive IBSs trams on a single id but is there a way to receive data from that id properly ?

juleq, 2020-05-18:
Yes, but that is probably not the answer that you want to hear: You can use two CAN busses.

Another would be to add a CAN to CAN gateway for one of the IBS to remap its messages.

The only other way IMHO is to force the IBS supplier to support multiple of his nodes on one bus, e.g. by offering to set a base id.

juleq, 2020-05-18:
And if you connect two nodes transmitting the same id to the same bus without further coordination, you need to consider the hint of @sebi2k1 : After a while one of them will have its error counters high enough that it will stop transmitting. If you have not observed that yet it might be that the IBS resets its controller automatically (not a solution here).

TIB28, 2020-05-19:
I separated the bus for each battery i can get info from first battery but the second does not reply.

juleq, 2020-05-19:
Is your kcd still the one that you have posted above? Are you sure that you have the multiplexor fields right (mux group count)? Have you verified the bus activity looks like you expect?

I assume that the multiplex has changed in the meantime because the batteries cannot agree on a common mux strategy. Mux is only for one node.

You could post your updated kcd. And you could post a short message trace.

TIB28, 2020-05-19:
Well my trams could be read by kayak and it works absolutely fine since the beginning but with this library there is not way to read multiplexed message correctly. So i think my multiplexor fields are right.

Here is my updated kcd file with 2 buses added: 
```
<!--
	This file is part of Kayak.

	Kayak is free software: you can redistribute it and/or modify
	it under the terms of the GNU Lesser General Public License as published by
	the Free Software Foundation, either version 3 of the License, or
	(at your option) any later version.

	Kayak is distributed in the hope that it will be useful,
	but WITHOUT ANY WARRANTY; without even the implied warranty of
	MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
	GNU General Public License for more details.

	You should have received a copy of the GNU Lesser General Public License
	along with Kayak.  If not, see <http://www.gnu.org/licenses/>.

-->
<NetworkDefinition xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://kayak.2codeornot2code.org/1.0" xsi:noNamespaceSchemaLocation="Definition.xsd">
	<Document name="kcd" version="1.0" .....">
	</Document>

	<Bus name="Main" baudrate="125000">

		…….

		<Message id="0x206" name="IBS">
				<Multiplex name="Mux" offset="0" length="16">
					<MuxGroup count="01">
						<Signal name="CourantB1" offset="16" length="16" endianess="big">
							<Value slope="0.1" intercept="-2000" unit="A"/>
						</Signal>
						<Signal name="UID_IBS_B1" offset="32" length="16" endianess="big"/>
						<Signal name="PourcentageB1" offset="48" length="8">
						<Value slope="1" unit="%"/>
						</Signal>
						<Signal name="SOH_B1" offset="56" length="8"/>
					</MuxGroup>
					<MuxGroup count="257">
						<Signal name="CourantB2" offset="16" length="16" endianess="big">
							<Value slope="0.1" intercept="-2000" unit="A"/>
						</Signal>
						<Signal name="UID_IBS_B2" offset="32" length="8" endianess="big"/>
						<Signal name="PourcentageB2" offset="48" length="8">
							<Value slope="1" unit="%"/>
						</Signal>
						<Signal name="SOH_B2" offset="56" length="8"/>
					</MuxGroup>
					<MuxGroup count="33">
						<Signal name="Capa_B1" offset="16" length="8"/>
						<Signal name="Type B1" offset="40" length="8">
							<LabelSet>
								<Label name="Liquide" value="0"/>
								<Label name="AGM" value="1"/>
								<Label name="Gel EFB" value="2"/>
						</LabelSet>
						</Signal>
					</MuxGroup>
					<MuxGroup count="289">
						<Signal name="Capa_B2" offset="16" length="8"/>
						<Signal name="Type B2" offset="40" length="8">
							<LabelSet>
								<Label name="Liquide" value="0"/>
								<Label name="AGM" value="1"/>
								<Label name="Gel EFB" value="2"/>
						</LabelSet>
						</Signal>
					</MuxGroup>
					<MuxGroup count="17">
						<Signal name="TemperatureB1" offset="16" length="8">
							<Value slope="1" intercept="-40" unit="°C"/>
						</Signal>
					</MuxGroup>
					<MuxGroup count="273">
						<Signal name="TemperatureB2" offset="16" length="8">
							<Value slope="1" intercept="-40" unit="°C"/>
						</Signal>
					</MuxGroup>
				</Multiplex>
			<Signal name="Info IBS" offset="0" length="16"/>
		</Message>
	</Bus>
	<Bus name="Battery1" baudrate="125000">
		<Message id="0x206" name="IBS">
				<Multiplex name="Mux" offset="0" length="16">
					<MuxGroup count="01">
						<Signal name=“AmpB1" offset="16" length="16" endianess="big">
						</Signal>
						<Signal name="UID_IBS_B1" offset="32" length="16" endianess="big"/>
						<Signal name="PercentageB1" offset="48" length="8">
						<Value slope="1" unit="%"/>
						</Signal>
						<Signal name="SOH_B1" offset="56" length="8"/>
					</MuxGroup>
					<MuxGroup count="17">
						<Signal name="TemperatureB1" offset="16" length="8">
							<Value slope="1" intercept="-40" unit="°C"/>
						</Signal>
					</MuxGroup>
				</Multiplex>
		</Message>
	</Bus>
	<Bus name="Battery2"  baudrate="125000">
		<Message id="0x206" name="IBS">
				<Multiplex name="Mux" offset="0" length="16">
					<MuxGroup count="257">
						<Signal name=“AmpB2" offset="16" length="16" endianess="big">
						</Signal>
						<Signal name="UID_IBS_B2" offset="32" length="8" endianess="big"/>
						<Signal name="PerrcentageB2" offset="48" length="8">
							<Value slope="1" unit="%"/>
						</Signal>
						<Signal name="SOH_B2" offset="56" length="8"/>
					</MuxGroup>
					<MuxGroup count="273">
						<Signal name="TemperatureB2" offset="16" length="8">
							<Value slope="1" intercept="-40" unit="°C"/>
						</Signal>
					</MuxGroup>
				</Multiplex>
		</Message>
	</Bus>
</NetworkDefinition>

```

juleq, 2020-05-19:
I assume when you say trams you mean frames?

Are battery 1 and 2 identical devices? If so, could you explain why battery 2 should set multiplex values differently than battery 1? How does it know that it is battery 2?

TIB28, 2020-05-19:
battery are not identical but it can be differentiate with MuxGroup count value

sebi2k1, 2020-05-19:
Can you give example frames?

sebi2k1, 2022-11-06:
no feedback

closed 2022-11-06 by sebi2k1