sebi2k1/node-can #89: Won't build on node:lts-alpine image

fabdrol, 2020-11-17:
I'm not entirely sure why this fails, but this image fails to build on the official `node:lts-alpine` image. 

Dockerfile:
```
FROM node:lts-alpine

# Install dependencies
RUN apk add --update --no-cache git gcc g++ python3 make bash nano curl wget

# Copy files
COPY . /opt/.essense-ai/reporter

# Remove uncessary files
RUN rm /opt/.essense-ai/reporter/Dockerfile; exit 0;
RUN rm /opt/.essense-ai/reporter/Makefile; exit 0;
RUN rm /opt/.essense-ai/reporter/package-lock.json; exit 0;

# Install NPM dependencies
RUN cd /opt/.essense-ai/reporter && npm install

# Start program
CMD ["/bin/bash", "-c", "/opt/.essense-ai/reporter/start.sh"]
```

Log:
```
Step 7/8 : RUN cd /opt/.essense-ai/reporter && npm install
 ---> Running in 1ae360469c77
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated to-iso-string@0.0.2: to-iso-string has been deprecated, use @segment/to-iso-string instead.
npm WARN deprecated mkdirp@0.3.0: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.

> @serialport/bindings@2.0.8 install /opt/.essense-ai/reporter/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=14.15.1 runtime=node arch=x64 libc=musl platform=linux)
make: Entering directory '/opt/.essense-ai/reporter/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:78:69: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
   78 |   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
      |                                                                     ^~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:135:71: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  135 |   uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
      |                                                                       ^~~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:175:70: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  175 |   uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);
      |                                                                      ^~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:215:70: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  215 |   uv_queue_work(uv_default_loop(), req, EIO_Flush, (uv_after_work_cb)EIO_AfterFlush);
      |                                                                      ^~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:270:68: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  270 |   uv_queue_work(uv_default_loop(), req, EIO_Set, (uv_after_work_cb)EIO_AfterSet);
      |                                                                    ^~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Get(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:314:68: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  314 |   uv_queue_work(uv_default_loop(), req, EIO_Get, (uv_after_work_cb)EIO_AfterGet);
      |                                                                    ^~~~~~~~~~~~
../src/serialport.cpp: In function 'void EIO_AfterGet(uv_work_t*)':
../src/serialport.cpp:329:96: error: no matching function for call to 'v8::Object::Set(v8::Local<v8::String>, Nan::imp::FactoryBase<v8::Boolean>::return_t)'
  329 |     results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(data->cts));
      |                                                                                                ^
In file included from /root/.cache/node-gyp/14.15.1/include/node/node.h:67,
[... 116 lines cut ...]
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 4.19.76-linuxkit
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/.essense-ai/reporter/node_modules/socketcan
gyp ERR! node -v v14.15.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @serialport/bindings@2.0.8 (node_modules/@serialport/bindings):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! socketcan@2.8.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the socketcan@2.8.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-17T10_22_37_608Z-debug.log
The command '/bin/sh -c cd /opt/.essense-ai/reporter && npm install' returned a non-zero code: 1
make: *** [docker_build] Error 1
```

dogmatic69, 2020-11-17:
There is an error about not finding the can header files, possibly gcc related, possibly `COPY` related. When it's nodegyp it is almost always missing parts of the gcc tool chain.

juleq, 2020-11-17:
Hi there. It seems that the linux kernel headers are either not installed (check Alpine pkg manager) or not visible to the build process (check your include path).

Offending:
 #include <linux/can.h>

That is the linux header for the socketcan API.

BR

sebi2k1, 2020-11-19:
Please install the Linux kernel headers package too. I believe it is called linux-headers on Alpine. E.g. https://pkgs.alpinelinux.org/package/edge/main/x86/linux-headers

closed 2020-11-19 by sebi2k1