sebi2k1/node-can #92: Multi-packet

juliomiguel1, 2020-12-17:
HELLO,

First of all, thanks for this node.

I have a small question:
Can you decode multi-packets with this library?

And if so, will you have an example?

juleq, 2020-12-17:
Hi.
What do you mean? There are several protocols for multi-frame transmissions. Which one are you interested in?

AFAIK this project does not have implementations of higher level protocols, but depending on your requirements it could be fairly straightforward to assemble raw messages yourself. But that really depends...

sebi2k1, 2021-01-11:
I did had some plans using ISOTP but there are so many different protocols out there it likely makes more sense to develop them in userspace.

Right now there are plans I is intended for signal based bus communication only.

closed 2021-01-11 by sebi2k1