sebi2k1/node-can #93: OSX compatibility

jbonnett92, 2021-01-17:
Hi,
When trying to install via NPM I get an "unsupported platform" error.
I also tried to build it on my machine and got `../src/rawchannel.cc:39:10: fatal error: 'linux/can.h' file not found`.

Can we add support for this please?

Thanks,
Jamie

juleq, 2021-01-17:
Hi,

I am a fellow user of this project.

This project implements support for socketcan in nodejs. Socketcan is a linux abstraction to allow a multitude of different interfaces to be used through just one API. To my knowledge, there is no equivalent on Mac and Win.

Supporting it would probably mean first developing the socketcan API in e.g. Mac OS user space. Further, the small amount of available device drivers on Mac would each require wrapping to that API accordingly.

Since the number of available interfaces with Mac support is small, I think just selecting a popular device and program a nodejs package against its driver (similar to this project, but not against socketcan) might be a faster (but less flexible) solution.

Here is some context from the guy that already implemented two interface drivers for the Mac. Thy are x86_64 but since they are user space they probably support M1 (have not tried yet.

https://github.com/mac-can/RusokuCAN/blob/master/Sources/MacCAN_Core/README.md

Best regards

jbonnett92, 2021-01-17:
Hi,
@juleq I have seen it working in Windows though...

sebi2k1, 2021-01-17:
As @juleq mentioned this project wraps the SocketCAN API which is only available on Linux.

If there is a standardized API on other platforms we can support it. 

Interfaces via Serial are usually highly proprietary and thus not supported as well.

closed 2021-01-17 by sebi2k1