sebi2k1/node-can #94: Error in installing in raspbian

abhibhatia98, 2021-03-11:
Hi @sebi2k1  I tried to install this library and it fails wit following log`gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/12.21.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
make: Entering directory '/home/pi/node_modules/node-red-contrib-canbus/node_modules/socketcan/build'
  CXX(target) Release/obj.target/can/src/rawchannel.o
../src/rawchannel.cc:404:34: error: ‘v8::Handle’ has not been declared
   static bool ObjectToFilter(v8::Handle<v8::Object> object, struct can_filter *rfilter)
                                  ^~~~~~
../src/rawchannel.cc:404:40: error: expected ‘,’ or ‘...’ before ‘<’ token
   static bool ObjectToFilter(v8::Handle<v8::Object> object, struct can_filter *rfilter)
                                        ^
../src/rawchannel.cc: In static member function ‘static void RawChannel::Init(v8::Local<v8::Object>)’:
../src/rawchannel.cc:98:40: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
     constructor.Reset(tpl->GetFunction());
                                        ^
In file included from /tmp/.node-gyp/12.21.0/include/node/node.h:67,
                 from ../../../../nan/nan.h:56,
                 from ../src/rawchannel.cc:23:
/tmp/.node-gyp/12.21.0/include/node/v8.h:6126:46: note: candidate: ‘v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/tmp/.node-gyp/12.21.0/include/node/v8.h:6126:46: note:   candidate expects 1 argument, 0 provided
../src/rawchannel.cc:99:76: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
     exports->Set(Nan::New("RawChannel").ToLocalChecked(), tpl->GetFunction());
                                                                            ^
In file included from /tmp/.node-gyp/12.21.0/include/node/node.h:67,
                 from ../../../../nan/nan.h:56,
                 from ../src/rawchannel.cc:23:
/tmp/.node-gyp/12.21.0/include/node/v8.h:6126:46: note: candidate: ‘v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/tmp/.node-gyp/12.21.0/include/node/v8.h:6126:46: note:   candidate expects 1 argument, 0 provided
../src/rawchannel.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE RawChannel::New(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rawchannel.cc:166:45: error: no matching function for call to ‘v8::Value::ToString()’
     Nan::Utf8String ascii(info[0]->ToString());
                                             ^
In file included from /tmp/.node-gyp/12.21.0/include/node/node.h:67,
                 from ../../../../nan/nan.h:56,
                 from ../src/rawchannel.cc:23:
/tmp/.node-gyp/12.21.0/include/node/v8.h:2668:44: note: candidate: ‘v8::MaybeLocal<v8::String> v8::Value::ToString(v8::Local<v8::Context>) const’
   V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
                                            ^~~~~~~~
/tmp/.node-gyp/12.21.0/include/node/v8.h:2668:44: note:   candidate expects 1 argument, 0 provided
In file included from /tmp/.node-gyp/12.21.0/include/node/v8-internal.h:14,
                 from /tmp/.node-gyp/12.21.0/include/node/v8.h:27,
                 from /tmp/.node-gyp/12.21.0/include/node/node.h:67,
                 from ../../../../nan/nan.h:56,
                 from ../src/rawchannel.cc:23:
/tmp/.node-gyp/12.21.0/include/node/v8.h:2684:31: note: candidate: ‘v8::Local<v8::String> v8::Value::ToString(v8::Isolate*) const’
                 Local<String> ToString(Isolate* isolate) const);
                               ^~~~~~~~
/tmp/.node-gyp/12.21.0/include/node/v8config.h:328:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
/tmp/.node-gyp/12.21.0/include/node/v8.h:2684:31: note:   candidate expects 1 argument, 0 provided
                 Local<String> ToString(Isolate* isolate) const);
                               ^~~~~~~~
/tmp/.node-gyp/12.21.0/include/node/v8config.h:328:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/rawchannel.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE RawChannel::AddListener(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rawchannel.cc:200:50: error: no matching function for call to ‘v8::Value::ToObject()’
         listener->handle.Reset(info[2]->ToObject());
                                                  ^
In file included from /tmp/.node-gyp/12.21.0/include/node/node.h:67,
                 from ../../../../nan/nan.h:56,
                 from ../src/rawchannel.cc:23:
/tmp/.node-gyp/12.21.0/include/node/v8.h:2672:44: note: candidate: ‘v8::MaybeLocal<v8::Object> v8::Value::ToObject(v8::Local<v8::Context>) const’
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
                                            ^~~~~~~~
/tmp/.node-gyp/12.21.0/include/node/v8.h:2672:44: note:   candidate expects 1 argument, 0 provided
[... 312 lines cut ...]
gyp ERR! node -v v12.21.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN notsup Unsupported engine for node-red-contrib-canbus@1.0.1: wanted: {"node":"=0.12.x"} (current: {"node":"12.21.0","npm":"6.14.11"})
npm WARN notsup Not compatible with your version of node/npm: node-red-contrib-canbus@1.0.1
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! socketcan@2.1.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the socketcan@2.1.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-11T11_08_58_791Z-debug.log
pi@raspberrypi:~ $
`

If you could provide any clue, I will be thankful

sebi2k1, 2021-03-14:
No idea, sorry. Looks like some issue with the node-dev headers in this environment. nodejs v12 is already supported and th CI builds on Xenial.

@juleq Ever seen this?

juleq, 2021-03-14:
Hi there,

I have never used node red, I always just install the socketcan package.

I have just checked into a pi, updated everything and npm installed socketcan. That resolved to socketcan@2.9.1 and it built just fine.

I remembered having seen an old version of the package under the name node-can. Npm installing that resolves to node-can@0.0.3-alpha.2. This does fail in a similar fashion as @abhibhatia98 attempt.

The log @abhibhatia98 has provided tries installing socketcan@2.1.3, which is fairly old.

@abhibhatia98 maybe asking the maintainer of the node-red-contrib-canbus package to update his socketcan dependency from 2.1.3 to 2.9.1 is the best way to fix your issue. I am not familiar with node red, but since the socketcan module API has been rather stable, maybe you can also update to the latest version yourself.

@sebi2k1 I vaguely remember that there has been a change to the v8 or nan API a while back that you have fixed back then. I guess it was somewhere between 2.1.3 and 2.9.1?

grodansparadis, 2021-03-14:
@abhibhatia98  You can try if [node-red-contrib-socketcan ](https://flows.nodered.org/node/node-red-contrib-socketcan) works better for you.

sebi2k1, 2021-03-16:
@juleq You are right, node-red-contrib-canbus (used by @abhibhatia98) includes socketcan-2.1.x only, which is too old for v12. Which was fixed by @tuna-f1sh after v.2.4.0.

Issue closed.

closed 2021-03-16 by sebi2k1