sebi2k1/node-can #97: Not working with Worker thread.

alexey11231, 2021-04-19:
If  "onMessage" listener is called inside worker thread "Segmentation fault (core dumped) " occurs upon received can msg.

sebi2k1, 2021-04-21:
You have a code snippet demonstrating the issue?

alexey11231, 2021-04-28:
**main:**
```
const { Worker, isMainThread } = require('worker_threads');

if (isMainThread) {
 
  new Worker("./worker.js");

}  
```

**worker.js:** 
```
 console.log("inThread")

var can = require('socketcan');

var channel = can.createRawChannel("can0", true);
 
channel.addListener("onMessage", function(msg) { console.log(msg); } );
 
channel.addListener("onMessage", channel.send, channel);

channel.start();`

```
**errormsg:** 
> 
> FATAL ERROR: HandleScope::HandleScope Entering the V8 API without proper locking in place
> Aborted
> 
 
 Today on first run it gave that msg. Before and after it was throwing segmentation fault.

juleq, 2021-04-28:
Seems one would need to litter the cpp code with multithreading protection to support workers. 

Could you share why you want to use a worker? The whole node idea is to eleminate multithreading complexity by offering an event driven infrastructure and non-blocking async io. The thread pools node uses for that is hidden under the hood.

alexey11231, 2021-04-29:
Nothing specific. I wanted to put can logic in separate worker, no good reason, just force of habit I guess.
I was just playing around and noticed this, so I decided to inform you about it. 

Btw great job with this(node-can), its quite useful. Ty !

juleq, 2021-04-29:
These flowers go straight to @sebi2k1 💐.

I have a direct comparison implementing stuff with node in one instance and C# and its TPL in another. I really appreciate the simplicity that the event driven approach provides for IO heavy use cases.

Your feedback regarding workers is definitely a limitation to keep in mind. If someone is interested in making himself familiar with thread safety mechanisms of V8, it could be a nice little project to implement support. But I have used node-can quite extensively in a busy embedded system and I never felt lonely without the thread safety 😃.

closed 2026-05-23 by sebi2k1