sebi2k1/node-can #99: serial port ?

ptorrent, 2021-06-08:
Hello,

Is it possible to user serial port with this library ?

juleq, 2021-06-08:
Do you mean serial to CAN interfaces? I doubt it. You would need a Linux socketcan driver that sits on your serial interface because this library does only support socketcan interfaces.

I assume it will be hard to find such a driver because these serial/CAN interfaces do not use standardized protocols on their serial side.

I can recommend Peak CAN interfaces (~200€). There might be cheaper options, just ensure that it ships with a socketcan driver (the Peak one does, I think they even are supported by mainline Kernels out of the box).

sebi2k1, 2021-06-08:
What about slcan? It’s a SocketCAN driver supporting serial devices but they have to support the SLCAN protocol.

https://github.com/torvalds/linux/blob/614124bea77e452aa6df7a8714e8bc820b489922/drivers/net/can/slcan.c

juleq, 2021-06-08:
Well. I guess it wan not _that_ hard to find 😋.

closed 2021-12-12 by sebi2k1