sebi2k1/node-can #100: Not getting correct value from a signal over 32bits length

touchlinux, 2021-12-10:
I got a callback from Signal.updateListeners when get message from CAN bus. The value of the signals were given correctly when the signals' length are 32 bits or less. But when the length is more than 32 bits, the values does not look accurate. I think the problem comes from the bit shift operation (<<) on https://github.com/sebi2k1/node-can/blob/master/socketcan.js#L344
For me, the bit shift works as circular shift.
console.log(1 << 32); // 1

sebi2k1, 2021-12-12:
Do you happen to have a snippet of a kcd file, input data and result? I’d like to extend the testcard to confirm a fix.

touchlinux, 2021-12-13:
The snippet of the kcd file I used looks like this.

	<Message id="0x011" name="IOReply" length="6">
		<Signal name="NumPinState" offset="0" length="48">
			<Value type="unsigned"/>
		</Signal>
	</Message>

I sent 2 messages to my node-can application over a CAN bus. They are
	$ cansend <can_if> 011#010000000000
	$ cansend <can_if> 011#010000000100

My application received 2 messages and logged as shown.

	{ id: 17, data: <Buffer 01 00 00 00 00 00> }
	Name  : NumPinState
	bitLen: 48
	bitOff: 0
	endian: little
	Value : 1

	{ id: 17, data: <Buffer 01 00 00 00 01 00> }
	Name  : NumPinState
	bitLen: 48
	bitOff: 0
	endian: little
	Value : 2

juleq, 2021-12-13:
Good thing you catched that.

Out of curiosity: I assume you have the state of 48 GPIOs in that signal? When I do IO extenders via CAN, I usually define one signal for each IO. Allows convenient usage by name and reduces bit twiddling. Any specific reason for not breaking the aggregate signal up?

touchlinux, 2021-12-13:
@juleq No, I don't have reason not to define each IO for individual signal. Actually your idea is a good suggestion for my application. But I am still thinking whether I need to use "Multiplex" for differentiating GPIOs. Either way, my problem is solved because I don't need to use more than 32 bits.

closed 2022-11-06 by sebi2k1