sebi2k1/node-can #102: Time to make some updates

daverathbone, 2022-05-20:
Please make some more simple send examples.
Exaplain CAN frame byte order.
I have been using this code for major uk vehicle company R&D for years but its hacking 
Why put stuff open source if you carnt tell the full story.

juleq, 2022-05-20:
I think that came off a little rude. This is offered free of charge and it can do a lot.

> Please make some more simple send examples.

Maybe clarify what your use case does look like. For example, you could write a few lines of pseudo code that show how you would like to use it and ask for comments.

> Explain CAN frame byte order...............

I can recommend reading up on the kcd format and comments on endianness and start bit conventions/conversions. It gets kinda messy when signals are big endian because different formats (e.g. dbc vs. kcd) use different conventions there. So byte order and bit numbering / reference bit are important to understand.

> I have been using this code for major uk vehicle company R&D for years now but its still just hacking 

🎶 Well, I've been to Hastings and I've been to Brighton 
I've been to Eastbourne too 
So what 🎶

> So many developers - have had a go but no one owns it.

I have developed a fairly heavy lifting multi-bus embedded logging application using this. And I was very grateful for the simplicity, the robustness, the in-code documentation and the responsiveness of @sebi2k1.

Further, I have drawn inspiration for own implementations I did in C# using the dbc format.

daverathbone, 2022-05-20:
Sorry if this came accross as rude.
But please point me to the documention of Socketcan?
Starting CAN......
Sorry this code does not support  please use teminal...GREAT!
sudo ip link set can1 type can bitrate 125000
sudo ifconfig can1 up

Sending A CAN frame ......
Sending a can frame  ID 100 with 8 bytes do this.........
var can = require('socketcan');
var channel1 = can.createRawChannel("can1", false /* ask for timestamps */);
var canmsg = { id: 100, data: Buffer.from([ 0, 0xFF, 0xFE, 0xFD, 0xFC, 0xFB, 0xFA, 0xF7 ]) };
channel1.start();
channel.send(canmsg);

juleq, 2022-05-20:
> Sorry if this came accross as rude.
> 
> But please point me to the documention of Socketcan?

The Linux socketcan API is described here. But this is just for completeness, it contains all the details that this project builds upon: https://www.kernel.org/doc/html/latest/networking/can.html

This project offers a simple API to use CAN interfaces (that support the Linux Socketcan API) in your Javascript application. I do not know of other documentation than the information provided on the Github project page. But there is in-code documentation. I can recommend reading the socketcan.js file. This will suffice to understand this projects API usage.

> Starting CAN......
> 
> Sorry this code does not support  please use terminal...GREAT!

This project requires that you start up your Linux socketcan interfaces yourself, which is very reasonable. You could choose to call the shell from Javascript to start them, but this can be a security concern because of the privileges required. You would probably have a startup script that sets up the interfaces, starts them, drops privileges and then starts the node.js app. The app could even be jailed.

> sudo ip link set can1 type can bitrate 125000
> 
> sudo ifconfig can1 up
> 
This is the way to setup your interfaces. You can google to find further examples and additional options, like auto-restart, etc.
> 
> Sending A CAN frame ......
> 
> Sending a can frame  ID 100 with 8 bytes do this.........
> 
> var can = require('socketcan');
> 
> var channel1 = can.createRawChannel("can1", false /* ask for timestamps if true*/);
> 
> var canmsg = { id: 100, data: Buffer.from([ 0, 0xFF, 0xFE, 0xFD, 0xFC, 0xFB, 0xFA, 0xF7 ]) };
> 
> channel1.start();
> 
> channel1.send(canmsg);
> 
Yes, this will send a raw message. 

What does not work for you?

sebi2k1, 2022-05-22:
Thanks @juleq for the support I couldn’t have said it better. 😀

I will close it for now.

closed 2022-05-22 by sebi2k1