sebi2k1/node-can #104: Install on node 16 failes

martijndierckx, 2022-07-30:
I'm trying to install socketcan on Ubuntu 22.04, node v16.16.0, node-gyp 9.0.0, python 3.10.4, gcc 11.2.0.
But I'm getting an error. 

Any ideas?

> npm ERR! code 1
npm ERR! path /home/can/node-992-canbus/node_modules/socketcan
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! make: Entering directory '/home/can/node-992-canbus/node_modules/socketcan/build'
npm ERR!   CXX(target) Release/obj.target/can/src/rawchannel.o
npm ERR! make: Leaving directory '/home/can/node-992-canbus/node_modules/socketcan/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.0.0
npm ERR! gyp info using node@16.16.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.10.4 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/can/node-992-canbus/node_modules/socketcan/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/can/.cache/node-gyp/16.16.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/home/can/.cache/node-gyp/16.16.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/home/can/.cache/node-gyp/16.16.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/home/can/node-992-canbus/node_modules/socketcan',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! make: g++: No such file or directory
npm ERR! make: *** [can.target.mk:113: Release/obj.target/can/src/rawchannel.o] Error 127
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:527:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Linux 5.15.0-43-generic
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /home/can/node-992-canbus/node_modules/socketcan
npm ERR! gyp ERR! node -v v16.16.0
npm ERR! gyp ERR! node-gyp -v v9.0.0
npm ERR! gyp ERR! not ok

martijndierckx, 2022-08-10:
Anyone have an idea?

juleq, 2022-08-10:
Not sure if the offending line says that g++ itself was not found or its input files. Maybe you can increase the verbosity somehow.

Is g++ in your path? Can you call it when opening a fresh shell without issue?

martijndierckx, 2022-08-10:
gcc was installed ... not g++. Installing it fixed the issue.
I got caught up on the ambiguous error message as well and assumed it was the file.

closed 2022-08-10 by martijndierckx

juleq, 2022-08-10:
Happy socketcanning then :).