sebi2k1/node-can #116: typescript error: Cannot find module '../build/Release/can.node'

technicallyfeasible, 2023-02-05:
I just upgraded to the latest version and am unable to compile my typescript app because of this error:

```
node_modules/socketcan/dist/socketcan.d.ts:1:22 - error TS2307: Cannot find module '../build/Release/can.node' or its corresponding type declarations.

1 import * as can from "../build/Release/can.node";
```

From what I can tell this is because the compiler cannot find the can.d.ts declaration because it is not part of the dist or release folders but remains in src.

Additionally, the exported type for "Message" is not the same as "Message" on RawChannel (which is not exported) so I am not able to properly type `rawChannel.send(message)` calls.

sebi2k1, 2023-02-05:
Can you share minimal example?

technicallyfeasible, 2023-02-05:
Here is a minimal example:
[socketcan_test.zip](https://github.com/sebi2k1/node-can/files/10610941/socketcan_test.zip)

I am running node 16.18.0

install node_modules with `yarn` and build using `yarn build`

sebi2k1, 2023-02-11:
@jbcpollak have you seen that before?

sebi2k1, 2023-02-11:
@technicallyfeasible using the tsconfig from socketcan directly "yarn build" runs through without any error.

jbcpollak, 2023-02-11:
hi apologies, I put my project down for a while - I'll take a look this weekend.

technicallyfeasible, 2023-02-11:
@sebi2k1 that's not really the point, is it? It's supposed to work with most typescript configurations, you can't require people to use a specific config just so that your library works.
I went through the options in the socketcan tsconfig.json and adding `"skipLibCheck": true,` will not report the error anymore. However, I would prefer  not adding that as it makes types in general less accurate.

sebi2k1, 2023-02-11:
Happy for another proposal how to solve it. I couldn’t find details yet. It’s a problem with the native modules but it’s not the location of the artificats.

jbcpollak, 2023-02-12:
Hi, I just whipped up an example project and was about to post it when I noticed I have `skipLibCheck` enabled on it. I'm out of time right now but will look into it and will see if I can remove it asap. here is the project in the meantime:

https://github.com/arcmotive/scutils-ts

jbcpollak, 2023-02-13:
I belief the above referenced PR will fix this! Apologies for the delay.

closed 2023-02-13 by sebi2k1

sebi2k1, 2023-02-13:
Will release a new version, please check if that works for you too.

technicallyfeasible, 2023-02-13:
This works for me without errors, thanks for the quick fix 👍

PR #117 (2023-02-13): Fix #116 by referencing ambient declarations
This change fixes #116 by referencing the ambient declarations used by the library itself. This ensures the TS declarations for the node-gyp files are exported by the library and available to downstream projects.

Apologies for the oversight!

Let me know if you have any questions about these changes.

Sidenote: you may want to run `lint` before the compile steps in your GitHub workflows - builds will be faster when there are linting errors (which typically come before the TS compilation errors).
commit 414664cc (2023-02-13): Fix #116 by referencing ambient declarations (#117)

* Fix #116 by referencing ambient declarations

* disable a few eslint rules

both of these rules are good in general, so I don't recommend disabling them, but in this case the triple-slash rule is incorrect for referencing .node files, and the `any` rule is too complicated to build a full type definition for just one use case.

---------