sebi2k1/node-can #126: PEAK System PCAN-USB with example

ptorrent, 2023-10-02:
Hello there !

THanks for the awesome job made !

Just have a question about example:

```
var can = require("socketcan");

var channel = can.createRawChannel("can0", true);

// Log any message
channel.addListener("onMessage", function(msg) { console.log(msg); } );

// Reply any message
channel.addListener("onMessage", channel.send, channel);

channel.start();
```

I'm using PEAK interface and it seems that **_can0_** is correctly mounted. When I start this example, I've no error but the script just close. Is that a normal behavior ?

Thanks for your support !

ptorrent, 2023-10-02:
Oh, seems ok now, just missing this 

 ip link set up can0

sebi2k1, 2023-10-02:
Thank you for the flower. Enjoy.

closed 2023-10-02 by sebi2k1