sebi2k1/node-can #127: Serial/TCP to CAN Proxy

ptorrent, 2023-10-02:
Hello !

I would like to create a TCP to CAN proxy. 

The data received over TCP is already in CAN format. What would be the best approach to achieve this? Is there a way with this library to directly write a buffer to can0 and listen to messages received from can0 as a buffer without any decoding process?

Seems that I'm only able to send/receive message as object

Thanks for your support :)

sebi2k1, 2023-10-02:
You are saying you receive a data stream on TCP as a sequence of CAN messages (id, length, data) and you want to generate CAN message based on that?

ptorrent, 2023-10-02:
I receive CAN message (buffer) on TCP and I would like to write this message (same buffer) on can0 interface.

Something like that

```
var channel = can.createRawChannel("vcan0", true);
channel.start();

var tcpClient;

net.createServer(function(client){
  tcpClient = client 
  tcpClient.on('data', x => channel.send(x))
})
channel.on('data', d => tcpClient.write(d))
```

sebi2k1, 2023-10-03:
That’s not as straight forward as you think.

CAN as a protocol is single message based, similar to USB. Frame size is limited by your hardware too. Standard CAN 2.0b support 8bytes max per frame while CANFD goes up to 64bytes per frame.

TCP, unlike UDP and CAN, is a streaming protocol. It takes care of breaking down your payload into single packets incl. retransmission if a packet got lost. CAN supports this only by additional protocol implementations (e.g. ISOTP or CANopens‘ SD frames). While ISOTP is supported by SocketCAN via a special module this extension only expose the raw CAN protocol.

So to overcome this, you have to define your own streaming-like protocol on top of CAN (it’s not that complicated). There are nonplans to wrap ISOTP in this extensions as well.

closed 2024-02-17 by sebi2k1