sebi2k1/node-can #130: Request: Return errno from native send()

poppfd, 2023-12-11:
The `RawChannel.send()` method returns -1 if the send cannot be completed.  Could the method somehow return an object including the `errno` set by the socket.h -> send() method?  Or better yet, return the string parsed by `strerror()` to translate the errno to the system specific error.

I was debugging a CAN interface issue and found the send() returned -1.  But I didn't realize I had a txqueuelen issue until I recompiled the socketcan C code to provide this errno context (`105   ENOBUFS   No buffer space available`).  It'd be nice if node-can could provide that error information.

sebi2k1, 2024-02-17:
We would have to introduce a new set of function otherwise we would break compatibility with existing code. Did you tried reading ERRNO directly from the script? The send-call itself is synchronous and this should be the only syscall in the call chain (except if nodejs is doing some hidden syscall).

poppfd, 2024-02-19:
Yea I understand backwards compatibility is a concern.  I'm not super familiar with natively compiled C libraries for node so I thought you might have a more elegant solution.
Not sure what you mean by "the script".  If you mean the node script, I did not think the C `errno` variable was available in that scope.  If I could get access to the `errno` variable, it looks like node will parse it with [util.getSystemErrorName(error.errno)](https://nodejs.org/api/util.html#utilgetsystemerrornameerr)
Like I said, I recompiled the `can.cc` file to return the errno string instead of number of bytes sent.  I found this to be a crude, but workable solution.

sebi2k1, 2024-02-26:
We can add an additional function to read the errno. Would that work?

poppfd, 2024-02-27:
Yea I think that would work.  So if we detect a failed send, we can read the `errno` and log from node.