sebi2k1/node-can #131: Nodejs 19 unable to return new can.RawChannel

SirJamesBrown, 2023-12-12:
Not entirely sure how to interpret this. 
I am on nodejs 19.9
both cjs and mjs give this apparent error on v4.0.4 and tried v4.0.1:
```
node_modules/socketcan/dist/socketcan.js:73
    return new can.RawChannel(channel, timestamps, protocol, false);
           ^

Error: Error while creating channel
    at Object.createRawChannel (/home/james/code/car/node_modules/socketcan/dist/socketcan.js:73:12)
``` 
for the following example
```
import * as can from 'socketcan'
// let can = require("socketcan")

let channel = can.createRawChannel('can0', true)
let msg = { id: 0x123, data: [0, 1, 2, 3, 4, 5, 6, 7] }

setInterval(() => {
  let out = {}
  out.id = 0x123
  out.data = new Buffer.from(msg.data)
  channel.send(out)
}, 100)

channel.start()
```

sebi2k1, 2023-12-13:
Is can0 available and up?

SirJamesBrown, 2023-12-13:
Yes all seems fine from what I can see.

SirJamesBrown, 2023-12-14:
So sorry to waste your time, was playing a bit more and realized I was not passing over ssh to the machine correctly, works like a charm.

closed 2023-12-14 by SirJamesBrown