sebi2k1/node-can #133: CAN bus error status

Kheirlb, 2024-03-12:
Is there a way for me to check bus status? ERROR_ACTIVE vs PASSIVE?

sebi2k1, 2024-04-07:
Nevertheless would be cool to have an API to retrieve the current bus state.

dbeckwith01, 2024-04-30:
@sebi2k1 Do you have an example on how to listen for the error state?  My can controller got into the `ERROR_ACTIVE` state and I'm trying get that in my application.

sebi2k1, 2024-05-01:
I don't have a hardware here with me to play around with, so I can give only some general advise.

Default state of your controller is error active, where it detects and reports error. If it detects to many issues it enters passive state (still transmit) and finally (depending on the error) bus off state (no rx/tx anymore).

Everytime the controller detects an error it raises the event to the host. In SocketCAN this is translated into an error frame (check if received frame has an 'err' member), if yes you can evaluate the exact error state by looking into the bytes of the frame. The mapping is defined in [error.h](https://elixir.bootlin.com/linux/v4.0/source/include/uapi/linux/can/error.h). The first byte for example tells you the current bus state of the controller.

By default we enabled all error types, but you can change that via setErrorFilters(), parameter is a bit mask of the error class (see error.h again).

I found a nice illustration of the behaviour of a CAN controller [online](https://www.csselectronics.com/pages/can-bus-errors-intro-tutorial#:~:text=Error%20Active%3A%20This%20is%20the,Error%20Flags'%20when%20detecting%20errors.).

closed 2024-10-12 by sebi2k1