sebi2k1/node-can #136: Seems not to work with node-js 22

MyHomeMyData, 2024-09-18:
I'm using socketcan in ioBroker adapter [e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan). Now I added node-js 22.x to the test-matrix. Test fails for socketcan, pls. have a look to the log:
https://github.com/MyHomeMyData/ioBroker.e3oncan/actions/runs/10885244181/job/30202383332

juleq, 2024-09-19:
From what I see it looks like an issue upstream:

https://github.com/nodejs/nan/issues/936

Could be wrong but there are suggestions to a workaround you could try to confirm.

MyHomeMyData, 2024-10-10:
When I clone `node-can` and do a `npm ci` it fails due to a problem within `nan.h`.

The cause seems to be a reference to **nan version 2.17.0** in `package-lock.json`.

After changing `"nan": "*"` to `"nan": ">=2.21.0"` in `package.json` and `npm install` now `npm ci` runs without error.

Would you please check?

sebi2k1, 2024-10-12:
Thanks for the hint. Tested it and fixed with 4.0.5.

closed 2024-10-12 by sebi2k1