sebi2k1/node-can #140: Module did not self-register

nemocazin, 2025-05-06:
I'm currently using the Electron framework, and I want to use SocketCAN for a feature.
Except that when running the app, I have the Javascript Error : Module did not self-register: '<path>/node_modules/socketcan/build/Release/can.node'.
I tried deleting all the node_modules directory and installing again all the package, did a rebuild, but none of the answers I can get on the net is helping me.

`
[electron] A JavaScript error occurred in the main process
[electron] Uncaught Exception:
[electron] Error: Module did not self-register: '<path>/node_modules/socketcan/build/Release/can.node'.
[electron]     at process.func [as dlopen] (node:electron/js2c/node_init:2:2214)
[electron]     at Module._extensions..node (node:internal/modules/cjs/loader:1356:18)
[electron]     at Object.func [as .node] (node:electron/js2c/node_init:2:2214)
[electron]     at Module.load (node:internal/modules/cjs/loader:1126:32)
[electron]     at Module._load (node:internal/modules/cjs/loader:967:12)
[electron]     at c._load (node:electron/js2c/node_init:2:13672)
[electron]     at Module.require (node:internal/modules/cjs/loader:1150:19)
[electron]     at require (node:internal/modules/cjs/helpers:119:18)
[electron]     at Object.<anonymous> (<path>/node_modules/socketcan/dist/socketcan.js:51:26)
[electron]     at Module._compile (node:internal/modules/cjs/loader:1271:14)
`

sebi2k1, 2025-05-07:
I am not sure how this is related to SocketCAN. We are not supporting electron officially. Any resources?

pazzazzo, 2025-06-07:
Downgrade electron to 31.2.0
if it still doesn't work:
- Install electron-rebuild as dev-dependencie,
- Add this script: `"postinstall": "electron-rebuild -f -w socketcan"`
- Remove node_modules and package-lock.json
- Run `npm install`

closed 2025-06-24 by sebi2k1