sebi2k1/node-can #141: Error: Module did not self-register

boezex, 2025-08-22:
Hello all,

I'm experiencing the same problem as nemocazin. This is on "regular" Node.JS on X86_64. Node version V24.1.0. I tried Node Version V20.19.4 and V18.20.8 as well, without luck. Tried rebuilding with npm rebuild socketcan, and removed the node_modules folder and package_lock.json. Then installed with npm install --rebuild-from-source socketcan.

Can we together find and fix this problem?

Thanks in advance.

`Can Module err: Error: Module did not self-register: '/home/app/node_modules/socketcan/build/Release/can.node'.
    at Object..node (node:internal/modules/cjs/loader:1925:18)
    at Module.load (node:internal/modules/cjs/loader:1469:32)
    at Module._load (node:internal/modules/cjs/loader:1286:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:235:24)
    at Module.require (node:internal/modules/cjs/loader:1491:12)
    at require (node:internal/modules/helpers:135:16)
    at Object.<anonymous> (/home/app/node_modules/socketcan/dist/socketcan.js:51:26)
    at Module._compile (node:internal/modules/cjs/loader:1734:14)
    at Object..js (node:internal/modules/cjs/loader:1899:10)`

MyHomeMyData, 2025-08-24:
I'm experiencing the same problem with adapter [ioBroker.e3oncan](https://github.com/MyHomeMyData/ioBroker.e3oncan/tree/main):
* adapter is working fine when installed on node.js 20 as well when installed on node.js 22
* however, when I do an upgrade from node.js 20 to 22, start of adapter failes permanently.
* node.js reports: `Error: Module did not self-register: '/opt/iobroker/node_modules/socketcan/build/Release/can.node'`

`npm rb && npm i` does fix the problem. However, that's not applicable for end user. Would you please check and fix? Thank you!

Here's the log:
```
2025-08-24 10:22:14.242  - error: host.vb-mint Caught by controller[0]:   code: 'ERR_DLOPEN_FAILED'
2025-08-24 10:22:14.242  - error: host.vb-mint Caught by controller[0]: }
2025-08-24 10:22:14.242  - error: host.vb-mint Caught by controller[0]: Node.js v22.18.0
2025-08-24 10:22:14.242  - error: host.vb-mint instance system.adapter.e3oncan.0 terminated with code 1 (JS_CONTROLLER_STOPPED)
2025-08-24 10:22:14.242  - info: host.vb-mint Restart adapter system.adapter.e3oncan.0 because enabled
2025-08-24 10:22:23.342  - info: host.vb-mint "system.adapter.e3oncan.0" disabled
2025-08-24 10:22:44.886  - error: host.vb-mint Caught by controller[0]: node:internal/modules/cjs/loader:1846
2025-08-24 10:22:44.886  - error: host.vb-mint Caught by controller[0]:   return process.dlopen(module, path.toNamespacedPath(filename));
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:                  ^
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]: Error: Module did not self-register: '/opt/iobroker/node_modules/socketcan/build/Release/can.node'.
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at Object..node (node:internal/modules/cjs/loader:1846:18)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at Module.load (node:internal/modules/cjs/loader:1423:32)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at Function._load (node:internal/modules/cjs/loader:1246:12)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at TracingChannel.traceSync (node:diagnostics_channel:322:14)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at wrapModuleLoad (node:internal/modules/cjs/loader:235:24)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at Module.require (node:internal/modules/cjs/loader:1445:12)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at require (node:internal/modules/helpers:135:16)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at Object.<anonymous> (/opt/iobroker/node_modules/socketcan/dist/socketcan.js:51:26)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at Module._compile (node:internal/modules/cjs/loader:1688:14)
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:     at Object..js (node:internal/modules/cjs/loader:1820:10) {
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]:   code: 'ERR_DLOPEN_FAILED'
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[0]: }
2025-08-24 10:22:44.887  - error: host.vb-mint Caught by controller[1]: Node.js v22.18.0
2025-08-24 10:22:44.887  - error: host.vb-mint instance system.adapter.e3oncan.0 terminated with code 1 (JS_CONTROLLER_STOPPED)
2025-08-24 10:22:44.887  - info: host.vb-mint Do not restart adapter system.adapter.e3oncan.0 because disabled or deleted
```

sebi2k1, 2025-08-28:
Can you check if it works with the upgraded NAN module?

boezex, 2025-08-29:
Yes this fixed the issue! Thank you very much. I also noticed that the module doesn't like worker_threads. Is there any probability of this being implemented?

closed 2025-08-29 by boezex